google-deepmind/mujoco #1031: Adding visual/debug objects visible in viewer?

manavkulshrestha, 2023-08-24:
**How to ask for help**

First, read our quick guide to
[asking good questions](https://github.com/deepmind/mujoco#asking-questions).
Below is a template for you to use:

Hi,

I'm a student and I'm trying to use MuJoCo for a research project in robotics.I'm looking for some help with adding visual objects to the simulator on the fly with code. This is mostly just for debugging/sanity check/visual purposes like wanting to maybe highlight a target location or check coordinates and things like that. From another issue I have gathered that physics items cannot be spawned without recompiling the model.

In the [tutorial notebook](https://colab.research.google.com/github/deepmind/mujoco/blob/main/python/tutorial.ipynb#scrollTo=Z6NDYJ8IOVt7), I saw that one example adds a visual capsule but this is associated with the renderer which I do not have (I'm just using a passive viewer).

Any guidance using this would be helpful

saran-t, 2023-08-25:
This isn't possible currently but I'm thinking about ways to implement this.

manavkulshrestha, 2023-08-25:
Is there any workaround or hacky way to achieve the desired effect?

kevinzakka, 2023-08-27:
Closing to consolidate into #1023.

closed 2023-08-27 by kevinzakka

saran-t, 2023-10-10:
Resolved in 2e15574b58df4bfda59e1642cf67699cb0a58667.