google-deepmind/mujoco #1032: Body Tree utility in MuJoCo (python)

joonhyung-lee, 2023-08-25:
Hi,

I'm a graduate student and I'm trying to use MuJoCo for my own robotics research.

I'm looking for some help with **Element Tree** function(utility).

Here is a screenshot, illustrating my question:
![Uploading Screenshot from 2023-0
[image]
8-25 22-42-10.png…]()
[image]
https://www.youtube.com/watch?v=e0babIM8hBo

When I press the **'`'** key, it turns on something called **Body Tree**, where can I find the python code for this?
I ask this in order to get the region occupied by each geometry created in the mujoco world.

I appreciate your guidance on this matter, and any insights you can provide would be greatly beneficial to my project.

Thank you!

quagla, 2023-09-28:
Hi @joonhyung-lee you can find the bounding box of each geom in `model.geom_aabb`. This is in the geom frame, so you will need to transform it to the world frame using `data.geom_xpos` and `data.geom_xmat`. Transform the first 3 numbers of `geom_aabb` (its center, which is **not** necessarily the same as `geom_pos`) then add the other three numbers (its half-sizes) multiplied by each column of `xmat` (the orientation of the box).

closed 2023-09-28 by quagla

joonhyung-lee, 2023-10-02:
Thanks for the details, I really appreciate it.