google-deepmind/mujoco #1033: Flickering when transparent meshs overlap

kim135797531, 2023-08-26:
Hello,

I've got this flickering issue when I render scenes in MuJoCo where transparent Meshs are overlapping. (on Ubuntu 20.04 with MuJoCo 2.3.7)

[image]

I've attached a minimal XML file that shows the problem.

<details>
  <summary>minimal XML</summary>

```XML
<mujoco>
    <visual>
        <headlight ambient="0.4 0.4 0.4" diffuse="0.8 0.8 0.8" specular="0.1 0.1 0.1"/>
    </visual>
    <worldbody>
        <body>
            <geom pos="0.02 0 0" name="sphere1" type="sphere" size="0.1" rgba="0.8 0.6 0.4 0.8"/>
            <freejoint/>
        </body>
        <body>
            <geom pos="-0.02 0 0" name="sphere2" type="sphere" size="0.1" rgba="0.8 0.6 0.4 0.8"/>
            <freejoint/>
        </body>
    </worldbody>
</mujoco>
```

</details>

Did some digging and it looks like the issue might be tied to limitations in OpenGL's glBlend function. (around line 1209 in render_gl3.c).

I've Googled around and saw that algorithms like Depth Peeling might fix this, but I'm not really an OpenGL expert so implementing it is a bit out of my league.

Any chance this can be sorted by tweaking the visual options in MuJoCo?

Thank you in advance.

yuvaltassa, 2023-08-26:
Unfortunately this is indeed due to a limitation of MuJoCo's built in renderer which uses an old version of openGL. We are working on connecting MuJoCo to more advanced renderers, which will solve this problem. 

A solution which will work now is to use Unity for rendering via our Unity plugin.

If you'd like to work on upgrading the built in renderer or improving the interface to Unity, please get in touch.

closed 2023-08-26 by yuvaltassa

kim135797531, 2023-08-27:
Thank you for the quick response. For now, I'll just continue using it as is.
I really appreciate the ongoing efforts in contributing to and maintaining the MuJoCo community!

kevinzakka, 2023-09-05:
Sweet model btw! Curious to know what exactly you are working on :-)

kim135797531, 2023-09-05:
@kevinzakka

Thank you for your interest in my model! If you're curious about the details, I'd recommend checking out my IEEE ICDL conference paper titled "[Simulating a Human Fetus in Soft Uterus](https://ieeexplore.ieee.org/document/9962201)". Unfortunately, I won't be able to release the model in the near future.