google-deepmind/mujoco #1035: Error Importing MuJoCo Scene in Unity

michaeljrose, 2023-08-29:
Hi,

I'm a student and I'm trying to use the MuJoCo Unity plug-in and HAPTIX to study human performance of referred robotic arm control.

I'm looking for some help with simply importing a model (MPL.xml for example) in Unity. I have the plug-in directory, native MuJoCo library (version 2.3.7 in the appropriate location and named properly), and mjhaptix150 all downloaded and am able to add the packages.json file to Unity (packages loads as version 2.3.8, running on Unity 2022.3.8f1 on Windows 10) but when I go to import a MuJoCo scene from the Assests tab, I receive the following error message.

[image]

I've been following the same instructions that collaborators had success with in a similar project and have tried importing models with older versions of the packages.json file and older versions of Unity to no avail, so I was wondering if anyone knows how to get passed this error message?

Thanks,
Michael

kevinzakka, 2023-08-29:
Did you try doing what the error message is suggesting you do? Older models might not load in newer MuJoCo versions.

Balint-H, 2023-08-30:
Hello, I recommend updating the imported XML to the newer version of MuJoCo by editing the xml, defining either autolimits, or setting limited="true"  in the defaults for tendons.

michaeljrose, 2023-08-30:
Thank you both for the comments, I tried to edit the XML code in a couple different ways -- either removing the range fields from the two tendons where it appears or by adding limited="true" to those two lines -- but am getting a different error message when I try to play it. I'm not sure where the parameter "native" is as it isn't in the MPL.xml file...

I'm also not super experienced with this type of coding so apologies if it's something obvious

[image]

michaeljrose, 2023-09-04:
It looks like these are the mesh files being loaded, the standard ones from the mjhaptix150/model/MPL/mesh folder, which are .STL files. This folder is called out in the console as the meshes locations source
[image]

closed 2024-07-03 by michaeljrose