google-deepmind/mujoco #1040: Unexpected contact behavior between box (as floor) and cylinder

garygra, 2023-09-05:
Hi,

I'm a student and I'm trying to use MuJoCo for research on motion planning and system identification.

I'm trying to simulate an environment where the floor has _patches_ with different friction coefficients. The vehicle should behave differently when traversing different patches. As the floor, I'm using boxes with `condim=6` and different friction, however I keep getting weird behavior when placing the robot near the limit of a box.

With the minimal XML model, the cylinder bounces off when it makes contact with the box. If the position of the floor changes from `pos="19.5 19.5 0"` to `pos="19.0 19.0 0"` the cylinder does not bounce. This also happens with `condim=3`.

Model:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco model="wheel-minimal">
  <compiler angle="radian"/>
  <size nconmax="100" njmax="500"/>
  <option integrator="implicit" timestep="0.01"/>
  <asset>
    <texture name='a' builtin="gradient" height="100" rgb1="1 1 1" rgb2="0 0 0" type="skybox" width="100"/>
    <texture name="texplane0" type="2d" builtin="checker" rgb1="0.14 0.83 0.43" rgb2="0.23 0.09 0.33" width="100" height="100" mark="cross" markrgb=".8 .8 .8"/>
    <material name="matplane0" reflectance="0.2" texture="texplane0" texrepeat="1 1" texuniform="true"/>
  </asset>
  <visual>
    <headlight ambient="0.6 0.6 0.6" diffuse="0.5 0.5 0.5" specular="0.2 0.2 0.2"/>
  </visual>
  <worldbody>
    <body name="floor" pos="0.0 0.0 0">
      <geom contype="1" friction="1 0 .01" name="floor" pos="19.5 19.5 0" euler="0 0 0" size="20 20 0.01" type="box" material="matplane0" condim="6"/> 
    </body>
    <body name="wheel-minimal" pos="0.0 0.0 0.6">
      <geom type="cylinder" friction="1 0 .01" contype="1" conaffinity="1" size=".5 0.2" euler="1.57 0 0" mass="0.5" condim="6"/>
      <joint name="free" type="free"/>
    </body>
  </worldbody>
</mujoco>
```

</details>

Here is a screenshot / video, illustrating my question:

https://github.com/deepmind/mujoco/assets/6044646/7906b076-4de2-4b79-843f-18115d6a4285

Is this behavior expected?
Do you have any alternatives or suggestions?

yuvaltassa, 2023-09-05:
Is the box geom very thin? Try making it thicker and report back.

garygra, 2023-09-06:
Making it thicker helps, it reduces how close it needs to be to the edge to trigger this behavior. 

In the attached video, the floor is significantly thicker, and only when the cylinder is 0.05 from the edge do I get the bounce.

https://github.com/deepmind/mujoco/assets/6044646/aaa4ec9b-8f89-4910-b04e-8474492a4e00

yuvaltassa, 2023-09-06:
Does increasing [mpr_iterations](https://mujoco.readthedocs.io/en/latest/XMLreference.html#option-mpr-iterations) help further?

garygra, 2023-09-06:
mpr_tolerance helps a bit. Thanks for the help.

closed 2023-09-18 by yuvaltassa