google-deepmind/mujoco #1041: Modify MjModel values during runtime(or just before mj_step)

nsk126, 2023-09-06:
Hi,

I'm a student and I'm trying to use MuJoCo for simulations on robot body dynamics.

I'm looking for some help with modifying the model obj values defined when
```
model = mujoco.MjModel.from_xml_path(path)
data = mujoco.MjData(model)
```
I intend to run simulations multiple times with few modified values model obj values. Following is my XML

<details>
  <summary>minimal XML</summary>

```XML
<mujoco>
  <worldbody>
  <body name="arm" pos="0 0 .01">
        <joint axis="0 0 1" limited="false" name="joint0" pos="0 0 0" type="hinge"/>
	<geom fromto="0 0 0 0.5 0 0" name="link0" rgba="0.0 0.9 0 1" size=".01" type="cylinder"/>
  </body>
  </worldbody>
</mujoco>
```

</details>

I am able to read the values i defined in xml by doing `model.geom('link0').pos`, but even though i modify this just before `mujoco.mj_resetData()`, it doesnt seem to update in the actual simulation or reflect in the rendering.

yuvaltassa, 2023-09-17:
This is surprising, but we can't diagnose without a MWE.

I suspect that you need to turn off `sameframe` which is an optimisation whereby the compiler tells the kinematics that a frame transformation can be skipped.

yuvaltassa, 2024-01-08:
Closing due to lack of activity.

closed 2024-01-08 by yuvaltassa