google-deepmind/mujoco #1042: How

Uptomylimit, 2023-09-06:
**How to ask for help**

First, read our quick guide to
[asking good questions](https://github.com/deepmind/mujoco#asking-questions).
Below is a template for you to use:

Hi,

I'm a (student / professor / engineer) and I'm trying to use MuJoCo for _____.

I'm looking for some help with ____.

Here is a model which explains my question:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco>
  <worldbody>
    <light pos=".4 -.4 .3" dir="-2 2 -1.5" diffuse=".6 .6 .6"/>
    <light pos="-.2 -.4 .3" dir="1 2 -1.5" diffuse=".6 .6 .6"/>
    <geom type="plane" size=".5 .5 .01"/>
    <body name="ball" pos="0 0 0.05">
      <freejoint/>
      <geom size=".03"/>
    </body>
    <body name="hammer" pos="-.05 0 0.18">
      <joint axis="0 1 0"/>
      <geom type="capsule" size=".01" fromto="0 0 0 -.15 0 0"/>
      <geom type="box" size=".02 .02 .03" pos="-.15 0 0"/>
    </body>
  </worldbody>
</mujoco>
```

</details>

Here is a screenshot / video, illustrating my question:

Uptomylimit, 2023-09-12:
I have solved this problem by setting mjmodel.jnt_range to fix part of joints. 

However, I find that the  freejoint doesn't have joint_range to limit itself. Thus, can I replace the freejoint with 3 slide joint and 3 hinge joint?

yuvaltassa, 2023-09-18:
A free joint is equivalent to 3 sliders and a ball, not 3 hinges.

Why don't you just remove the joints? Having joints with very tight ranges is not recommended.

closed 2023-09-18 by yuvaltassa

Uptomylimit, 2023-09-18:
> A free joint is equivalent to 3 sliders and a ball, not 3 hinges.
> 
> Why don't you just remove the joints? Having joints with very tight ranges is not recommended.

Because I want record the joint angles to initial my robot state.

Uptomylimit, 2023-09-19:
Thanks for your answer!

Now I can remove the joints by calculating the relative position and create a new xml model to initial my robot state!