google-deepmind/mujoco #1043: Best way to access images from multiple different named cameras using Python and visualize them in real-time?

jomian78, 2023-09-06:
I am currently building a mujoco environment to run multi-agent learning experiments. 

Each agent in the environment has its own camera. Every time I forward step the environment I would like each agent to save a camera image from its current location. 

Is there a good example of how to accomplish this using the native Python bindings?

I've experimented with running this two different ways:

1) with dm_control:
`
physics = mujoco.Physics.from_xml_path(xml_file)
physics.reset()
i = 0
num_steps = 100
cam1_imgs = []
cam2_imgs = []
while i < num_steps: 
    # dynamics code omitted
    cam1_img = physics.render(camera_id="cam1")
    cam1_imgs.append(cam1_img)
    cam2_img = physics.render(camera_id="cam2")
    cam2_imgs.append(cam2_img)
    
    physics.step()
    i += 1
`
2) with the native Python bindings:
`
model = mj.MjModel.from_xml_path(xml_path) 
renderer = mujoco.Renderer(model, 480, 640)
data = mujoco.MjData(model)

i = 0
num_steps = 100
cam1_imgs = []
cam2_imgs = []

while i < num_steps:
    mujoco.mj_forward(model, data)
    renderer.update_scene(data, camera="cam1")
    cam1_img = renderer.render()
    cam1_imgs.append(cam1_img)

    renderer.update_scene(data, camera="cam2")
    cam2_img = renderer.render()
    cam2_imgs.append(cam2_img)
`

However, both of these methods are fairly slow with even just 2 cameras, especially when we try to visualize the camera feeds (using glfw or matplotlib). 

Does anyone have any suggestions on a faster/more efficient way to access the camera images, and also visualize them in real-time to the user?

yuvaltassa, 2023-09-17:
You're using GPU rendering, yes?

In principle, adding a native multicamera render call should be pretty straightforward, if you can read render_gl3.c

I suspect @aftersomemath could do it rather quickly 😁

jomian78, 2023-09-19:
> You're using GPU rendering, yes?
> 
> In principle, adding a native multicamera render call should be pretty straightforward, if you can read render_gl3.c
> 
> I suspect @aftersomemath could do it rather quickly 😁

Correct, am using GPU rendering - I'll look into adding a native multicamera render call (if @aftersomemath or someone else has done this for a similar application I would also be interested in using it). Thanks!

closed 2023-09-19 by jomian78

alishbaimran, 2023-09-20:
Also wondering same thing, what solution did you go with?

yinkangning0124, 2024-04-10:
Hi, im wondering which version of mujoco is in the issue? I am currently using mujoco 2.2.2, it seems that mujoco 2.2.2 do not have Rendered() class