google-deepmind/mujoco #1045: Why simulation version 2.3.7 uses more than twice as much CPU as 2.3.6?

HuNingHe, 2023-09-07:
Just run the built `simulation`(version 2.3.6 and 2.3.7) at [Release](https://github.com/deepmind/mujoco/releases), and compare their CPU usage. On my computer, the 2.3.6 only takes 2% of CPU at most but 2.3.7 takes %6 of CPU by average. 

- Include the following context:
  - Operating system: Ubuntu 20.04 and Ubuntu 22.04
  - MuJoCo version 2.3.7
  - CPU:I7 6900HQ

yuvaltassa, 2023-09-07:
Do you have capacity to build both versions from source and see if you can replicate this locally?

If yes, you should be able to use standard profiling tools to figure out where the difference is coming from. There is nothing obvious in the [2.3.7 changelog](https://mujoco.readthedocs.io/en/2.3.7/changelog.html) that we can point to as a possible suspect.

HuNingHe, 2023-09-08:
> Do you have capacity to [build both versions from source](https://mujoco.readthedocs.io/en/2.3.7/programming/index.html#building-from-source) and see if you can replicate this locally?
>
> If yes, you [...]

I‘m sorry to post this issue, It seems that my laptop power is not connected. I will close this issue.

closed 2023-09-08 by HuNingHe