google-deepmind/mujoco #1046: Converting a V-Rep model, .ttt file(binary) to xml file

Gokul2411, 2023-09-08:
Hi,

I'm a student  and I'm trying to use MuJoCo for simulating a dvrk robot.

I'm looking for some help with converting a .ttt file to a mujoco based xml file. I have a .ttt file of the dvrk robot and wish to convert it to a .xml file so I can use it in Mujoco.

Please let me know if there is any preexisting methods to convert such files to a .xml file

Thank you

kevinzakka, 2023-10-02:
Is it possible for you to export the .ttt file as a URDF? I recall seeing this export functionality a few years ago when messing with Coppelia sim. If you can do that, then MuJoCo can load your URDF and convert it to a MJCF. Of course, you'll have to do additional cleaning up like adding actuators.

closed 2023-10-02 by kevinzakka

Gokul2411, 2023-10-02:
Okay, thanks for the information