google-deepmind/mujoco #1047: How can I visualize a simulation ran inside a docker container

BolunDai0216, 2023-09-09:
Hi,

I'm to use MuJoCo for running robotic simulations from a docker container on an M1 Macbook Pro. I'm looking for some help with visualizing the simulation in my host OS. 

To reproduce the steps I took first, I opened `XQuartz` inside my MacOS, and opened a terminal and typed

```bash
xhost +
```

which gave me the output

```bash
access control disabled, clients can connect from any host
```

then I built the following docker file

```dockerfile
FROM ubuntu:latest  

RUN apt-get update
RUN apt-get install -y x11-apps x11-utils libx11-dev
RUN apt-get install -y python3-pip
RUN apt-get install -y git
RUN python3 -m pip install mujoco

WORKDIR /home
RUN git clone https://github.com/BolunDai0216/PyMuJoCoBase.git

WORKDIR /home/PyMuJoCoBase
RUN python3 -m pip install -e .

WORKDIR /home/PyMuJoCoBase/PyMuJoCoBase/examples

CMD bash
```

Then, I used the following command to build a docker image

```bash
docker build -t gui .
```

and I started a docker container using the command

```bash
docker run -it --net=host -e DISPLAY=docker.for.mac.host.internal:0 -v /tmp/.X11-unix:/tmp/.X11-unix:ro gui:latest
```

I first tested if I can connect to the host X11 server by running 

```bash
xeyes
```

which gave me the following output

[image]

Then, I tried to run a simple mujoco simulation, which gave me the following error

```bash
root@docker-desktop:/home/PyMuJoCoBase/PyMuJoCoBase/examples# python3 example_control_pendulum.py
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
/usr/local/lib/python3.10/dist-packages/glfw/__init__.py:916: GLFWError: (65543) b'GLX: Failed to create context: BadValue (integer parameter out of range for operation)'
  warnings.warn(message, GLFWError)
/usr/local/lib/python3.10/dist-packages/glfw/__init__.py:916: GLFWError: (65538) b'Cannot set swap interval without a current OpenGL or OpenGL ES context'
  warnings.warn(message, GLFWError)
Traceback (most recent call last):
  File "/home/PyMuJoCoBase/PyMuJoCoBase/examples/example_control_pendulum.py", line 93, in <module>
    main()
  File "/home/PyMuJoCoBase/PyMuJoCoBase/examples/example_control_pendulum.py", line 87, in main
    sim = ContolPendulum(xml_path)
  File "/home/PyMuJoCoBase/PyMuJoCoBase/examples/example_control_pendulum.py", line 11, in __init__
    super().__init__(xml_path)
  File "/home/PyMuJoCoBase/PyMuJoCoBase/mujoco_base.py", line 30, in __init__
    self.context = mj.MjrContext(self.model, mj.mjtFontScale.mjFONTSCALE_150.value)
mujoco.FatalError: gladLoadGL error
```

running the same command inside MacOS gives me the following simulation

[image]

Any help with resolving the OpenGL issue in docker would be greatly appreciated!

Thanks in advance!

Xi-HHHM, 2024-05-05:
Hi, having a similar issue here. Have you found the solution?

BolunDai0216, 2024-05-05:
> Hi, having a similar issue here. Have you found the solution?

Unfortunately I still have no idea how to get it to work on MacOS.

BolunDai0216, 2024-06-29:
For these two lines

```bash
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
```

the problem can be solved by running

```bash
export LIBGL_ALWAYS_INDIRECT=1
sudo apt-get install -y mesa-utils libgl1-mesa-glx
```

as shown in this [post](https://unix.stackexchange.com/a/642418). However, I still do not know how to solve the problem of 

```bash
/usr/local/lib/python3.10/dist-packages/glfw/__init__.py:914: GLFWError: (65543) b'GLX: Failed to create context: BadValue (integer parameter out of range for operation)'
  warnings.warn(message, GLFWError)
ERROR: could not create window

Press Enter to exit ...
```

This somehow feels like is due to the lack of an NVIDIA GPU, which if it is the case is not something we can deal with on MacOS.