google-deepmind/mujoco #1050: ArgumentNullException: Value cannot be null when running a basic scene using the Unity plugin

ih, 2023-09-11:
Hi,

I'm a hobbyist and I'm trying to use MuJoCo for learning more about deep reinforcement learning and robotics.

I'm looking for some help with getting started and I'm trying to follow this [YouTube video](https://youtu.be/eyzzsGJ1iic?si=FIACAB0LPatOcfT8) creating a sphere that rolls down a plane.

I believe I've correctly installed the plugin following the [docs](https://mujoco.readthedocs.io/en/latest/unity.html) and after following the video to create a plane and sphere when I hit run I get the exception
```
System.ArgumentNullException: Value cannot be null.
Parameter name: native
  at (wrapper managed-to-native) System.Runtime.InteropServices.Marshal.copy_to_unmanaged_fixed(System.Array,int,intptr,int,void*)
  at System.Runtime.InteropServices.Marshal.Copy (System.Byte[] source, System.Int32 startIndex, System.IntPtr destination, System.Int32 length) [0x0001e] in <b89873cb176e44a995a4781c7487d410>:0 
  at Mujoco.MjVfs.AddFile (System.String filename, System.String contents) [0x00072] in C:\Projects\mujoco\unity\Runtime\Tools\MjVfs.cs:60 
  at Mujoco.MjEngineTool.LoadModelFromString (System.String contents) [0x0000e] in C:\Projects\mujoco\unity\Runtime\Tools\MjEngineTool.cs:68 
  at Mujoco.MjScene.CompileScene (System.Xml.XmlDocument mjcf, System.Collections.Generic.IEnumerable`1[T] components) [0x00001] in C:\Projects\mujoco\unity\Runtime\Components\MjScene.cs:179 
  at Mujoco.MjScene.CreateScene (System.Boolean skipCompile) [0x00143] in C:\Projects\mujoco\unity\Runtime\Components\MjScene.cs:171 
  at Mujoco.MjScene.Start () [0x00008] in C:\Projects\mujoco\unity\Runtime\Components\MjScene.cs:89 
```
I am using Unity 2022.3.7f1 and MuJoCo 2.3.7 with Windows 11 Pro (AMD Ryzen 7 3700X).

I'm familiar with Unity, but have never used MuJoCo.

Here is a model I exported of the scene:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco>
  <worldbody>
    <geom density="1000" type="plane" size="9.999999 9.999999 1" pos="0 0 0" quat="-0.9992292 0 -0.03925679 0" priority="0" contype="1" conaffinity="1" group="0" condim="3" solmix="1" solref="0.02 1" solimp="0.9 0.95 0.001" margin="0" gap="0" friction="1 0.005 0.0001" fluidshape="none" fluidcoef="5 0.25 1.5 1 1" name="GameObject_0" />
    <body pos="0 0 3.54" quat="-1 0 0 0" gravcomp="0" name="Sphere Body_1">
      <geom density="1000" type="sphere" size="0.5" pos="0 0 0" quat="-1 0 0 0" priority="0" contype="1" conaffinity="1" group="0" condim="3" solmix="1" solref="0.02 1" solimp="0.9 0.95 0.001" margin="0" gap="0" friction="1 0.005 0.0001" fluidshape="none" fluidcoef="5 0.25 1.5 1 1" name="Sphere_2" />
      <freejoint name="Free Joint_3" />
    </body>
  </worldbody>
  <contact />
  <tendon />
  <equality />
  <actuator />
  <sensor />
  <compiler coordinate="local" />
  <option gravity="0 0 -9.81" timestep="0.02" />
  <asset />
</mujoco>
```

</details>

Here are some screenshots:

[image]

[image]

I inspected the arguments to Marshal.Copy and none of them were null.

I'm not sure if this is a bug or if I'm missing a setup step or something like that. Thanks for any help!

Balint-H, 2023-09-21:
Could you expand the Sphere Body GameObject in the hierarchy to see what components were added? You also might be missing a MjGlobalOptions component from your scene.

Balint-H, 2023-09-21:
You can also export the Unity scene to XML to see what is used to run the Mujoco scene

Balint-H, 2023-09-21:
Could you try running it with the 2.3.7 release or earlier version of the plugin? Lastly, this error may sometimes be caused by a mismatch in DLL and plugin version.

ih, 2023-10-05:
Thanks for your suggestions and sorry for the delayed reply, I was traveling recently. 

I tried all the above and what worked was reverting the MuJoCo repo to the commit tagged 2.3.6.

Seems like maybe there is a bug, but I don't see an option to change the label on this issue. Let me know if it'd be helpful to create a new issue/close this one.

brownbrowny, 2023-10-06:
I get the same error message when using version 2.3.6. for a robot model when pressing play. Thus I switched back to version 2.3.5.

anayebi, 2023-10-06:
I also get this error with the latest version of the Mujoco github repo, but downgrading to the [release version of 2.3.7](https://github.com/google-deepmind/mujoco/archive/refs/tags/2.3.7.zip) works for me.