google-deepmind/mujoco #1053: Won't play in unity editor but Build and runs

calebjthomson, 2023-09-12:
XML 👍 
<mujoco>
	<include file="LukeHand.xml"/>

    <worldbody>
		<geom type="plane" material="MatPlane" pos=".6 0 0.1" zaxis="-1 0 1" size=".1 .5 .025"/>
		<geom type="plane" material="MatPlane" pos="-.6 0 0.1" zaxis="1 0 1" size=".1 .5 .025"/>
		<geom type="plane" material="MatPlane" pos="0 .6 0.1" zaxis="0 -1 1" size=".5 .1 .025"/>
		<geom type="plane" material="MatPlane" pos="0 -.6 0.1" zaxis="0 1 1" size=".5 .1 .025"/>

        <body name="obj1" pos="-0.2 0 .1">
            <geom type="box" rgba="0 1 0 1" size="0.05 0.03 0.05" density="500"/>
            <joint class="free"/>
        </body>

        
            
            
        

        <body name="obj3" pos="0.1 0 .025">
            <geom type="box" pos="-0.1 0 0" material="MatWoodG" size="0.01 0.3 0.1"/>
            
            
            
            
            
            
        </body>
    </worldbody>
</mujoco>

XML 2:
<mujoco model="LukeHand">
    <compiler angle="degree" meshdir="mesh/" texturedir="texture/" inertiafromgeom="auto"/>
    <option timestep="0.001" iterations="50" apirate="100" solver="Newton" cone="elliptic"/>
    <size njmax="600" nconmax="150" nstack="300000"/>
    
    <default>
        <position ctrllimited="true" forcelimited="true" gear="57.29577951"/>
        <geom solimp="0.95 0.99 0.001" solref="0.01 1" friction="1.0 0.01 0.001"/>

        <default class="robot">
            <geom material="robot" contype="1" conaffinity="2" condim="4"/>
            <site rgba=".3 .9 .3 .3" type="box" group="3"/>
            <joint limited="true" pos="0 0 0" axis="1 0 0" range="0 90" damping="0.1" armature="0.002"/>

            <default class="RP">
                <joint damping="0.05" armature="0.001"/>
            </default>
        </default>

        <default class="couple">
            <equality solref="0.05 0.5" solimp="0.0001 0.0001 0"/>
        </default>    

        <default class="free">
            <joint type="free" damping="0" armature="0" limited="false"/>
        </default>     
    </default>

    <statistic extent="0.4" center="0 -0.2 0.2"/>

    <visual>
        <quality shadowsize="2048"/>
        <map fogstart="1" fogend="3" znear="0.05" force="0.002" stiffness="200"/>
        <headlight diffuse=".6 .6 .6" specular="0 0 0" active="1"/>
        <scale contactwidth="0.1" contactheight="0.03" forcewidth="0.05"/>
    </visual>

[... 237 lines cut ...]
    <sensor>
        <actuatorpos actuator="wristRotation" cutoff="175"/>
        <actuatorpos actuator="wristFlexion"  cutoff="55"/>
        <actuatorpos actuator="thumbYaw"      cutoff="90"/>
        <actuatorpos actuator="thumbPitch"    cutoff="100"/>
        <actuatorpos actuator="index"         cutoff="90"/>
        <actuatorpos actuator="MRP"           cutoff="90"/>

        <touch site="palmProximal" cutoff="25.5"/>
        <touch site="palmDistal"   cutoff="25.5"/>
        <touch site="handDorsal"   cutoff="25.5"/>
        <touch site="handEdge"     cutoff="25.5"/>
        <touch site="thumbTip"     cutoff="25.5"/>
        <touch site="thumbRadial"  cutoff="25.5"/>
        
        
        <touch site="indexTip"     cutoff="25.5"/>
        <touch site="indexLateral" cutoff="25.5"/>
        <touch site="middleTip"    cutoff="25.5"/>
        <touch site="ringTip"      cutoff="25.5"/>
        <touch site="littleTip"    cutoff="25.5"/>
    </sensor>
</mujoco>

For whatever reason I can build an run this script from Unity but it doesn't allow me to press play an interact with the model in Unity.

calebjthomson, 2023-09-19:
Confirmed i'm having this same issue with other mujoco scenes and not just the XML files above... I'm running 2021.3.6f1

Balint-H, 2023-09-21:
Do you get any error messages in the console?

Balint-H, 2025-02-04:
Closing this for now, but please get in touch if you still have an issue you'd like help with.

closed 2025-02-04 by Balint-H