google-deepmind/mujoco #1054: Can I update the maximum number of assets in python?

guyazran, 2023-09-13:
Hi,

I'm an engineer and I'm trying to use MuJoCo for simulating physics of realistic scenes. I must dynamically create the scene XML and assets dictionary. I'm looking for some help with loading such scenes that have a large amount of assets using the python interface.

I have been using `mujoco.MjModel.from_xml_string(xml_string, assets)` to compile the scene and receive a MuJoCo model object. However, my new scenes are rich in detail and have over 8,000 assets. With these new scenes I get the error `assets dict is too big`. I know this is not a limitation of MuJoCo since I am able to save the model and assets to disk and load them with `mujoco.MjModel.from_xml_path` with no problem. Sadly, this solution is highly inefficient. Is it possible to update the maximum number of allowed assets when loading with `from_xml_string` through the python bindings?

--------------------------

Taking a deeper dive into the MuJoCo code, I see that the `assets dict is too big` error is thrown in the `LoadModelFileImpl` function defined in the file `mujoco/python/mujoco/structs.cc`. The error that leads to this is named `vfs_error` and is received via a call to the `mj_makeEmptyFileVFS` found in the file `mujoco/src/engine/engine_vfs.c`. In this function, an error occurs when the number of files in the VFS is too big, as can be seen in lines 110-12 of the file:
```cpp
if (vfs->nfile >= mjMAXVFS-1) {
    return 1;
}
```

I am assuming this VFS is derived from the assets dict and that `nfile` corresponds to the number of entries. Is this correct? If so, is it possible to update the `mjMAXVFS` constant through the python interface?

saran-t, 2023-09-14:
Is there a particular reason why you're using the assets dict rather than referring to paths on disk directly in the XML?

guyazran, 2023-09-14:
Yes. The scenes are generated dynamically and saving them to disk would be an unnecessary and expensive step. For performance purposes, I would much rather compile the model and assets that are already in memory than to save and reload them. Furthermore, I sometimes use dm_control's PyMJCF tool to combine the scenes with robots and other attachments, which internally calls `MjModel.from_xml_string(xml_string, assets)` anyway.

saran-t, 2023-09-14:
I'd really like to deprecate VFS as much as possible and replace it with the new mjpResourceProvider mechanism. Especially for Python users, this should make things much more flexible, but it will mean that you'll need to prefix all your asset "paths" with some string (not yet decided, probably something like `pyasset:somefilename.extension`. Would that work for you?

guyazran, 2023-09-14:
Thank you for your fast responses @saran-t.

Perhaps I'm missing the point of `[mjpResourceProvider](https://mujoco.readthedocs.io/en/stable/programming/extension.html#resource-providers)`, but it seems like we would still have to save the MJCF and reload it once again. Is there any way to avoid this entirely? Note that PyMJCF would also need to support this. otherwise we can't attach MJCFs together.

guyazran, 2023-09-26:
I was able to play around with  [mjpResourceProvider](https://mujoco.readthedocs.io/en/stable/programming/extension.html#resource-providers) in C++. I see now that there isn't actually any need to save any files, but only to prefix the resource paths (i.e. the "file" attributes in `<include/>` tags) like @saran-t mentioned [here](https://github.com/google-deepmind/mujoco/issues/1054#issuecomment-1719343816). This is a good fix for my issue. is there a way to access this feature via the python API?

saran-t, 2023-09-27:
Not currently, I'm afraid, but I'd be open to adding it to a future version.

saran-t, 2023-09-27:
If you're on Linux a possible workaround for the time being would be to mount a `tmpfs` and save your assets as files on it. This would amount to an overhead of one `memcpy` per asset which shouldn't be too bad (certainly no worse than what MuJoCo Python bindings would have to do anyway).

guyazran, 2023-10-02:
Thank you for this suggestion. I did not know about `tmpfs` and the concept of RAMDisk. Unfortunately, we require a cross-platform solution. It seems like the only way to do this currently in python is using [PyFilesystem](https://docs.pyfilesystem.org/en/latest/reference/memoryfs.html), but I did not find a way to monkey-patch this through to MuJoCo. Hopefully the resource provider mechanism [will be added soon](https://github.com/google-deepmind/mujoco/issues/1054#issuecomment-1737851052).

kbayes, 2023-10-02:
I’m the designer/ implementer of resource providers. I’ve been out for health reasons for the last month, but I should be back to work in a week or two, and I can address your issue. :) Sorry about the wait!

guyazran, 2023-10-04:
Thanks @kbayes. I hope you are feeling OK.

Seeing as this as I opened this issue as a question, should we migrate to a feature request?

yuvaltassa, 2023-10-04:
Yes please, let's make this a feature request issue.

closed 2023-10-04 by yuvaltassa