google-deepmind/mujoco #1055: C++ vs the python bindings

coder-rakeshpaul, 2023-09-14:
Hi,

I'm trying to use mujoco for some reinforcement learning task and I'm confused if I should stick to the python bindings or the c++ API I'm proficient in python and I have some knowledge of c++ but would ideally prefer sticking to python. the task that I'm planing will require some real time data as input and requires fast response while a small delay is still acceptable in case I use the python bindings. I would have implemented a small model and tested out but I'm still quite new to mujoco and struggling to get a good grip on it. The task is urgent and I don't have time to learn both the C++ and the python bindings.

thank you

breakds, 2023-09-18:
I am not sure what you meant by "require some real time data as input and requires fast response". MuJoCo is a simulator and I cannot imagine how it connects to "require some real time data". Do you run your RL training in simulation?

On the other hand, in terms of performance, running the simulation step with either python or c++ should have very similar speed. However, since you are running reinforcement training, inevitably you will need to implement observation and reward logic, as well as running a lot of simulation in parallel. The performance gap is pretty significant if you implement those logic in c++ vs python, just FYI.

coder-rakeshpaul, 2023-09-19:
I apologize for the confusion. The "requires some real time data"  is more specific to the project I working on and might not be relevant to mujoco. The second part of your answer is what I was looking for.

thank you for the response

closed 2023-09-19 by coder-rakeshpaul