google-deepmind/mujoco #1057: Changing body/geom position at runtime

manavkulshrestha, 2023-09-15:
Hi, I'm a student and I'm trying to use MuJoCo for model based robot control. I want some dynamic obstacles in the environment (moving walls/obstacles/humans etc.) that still have collision, but don't have complex movement models. That is, I want to have it just teleport to a series of locations to follow some path

I was able to teleport a site in this manner, and even a geom like so:
 <details>
  <summary>site and geom object inside world body</summary>

mjcf:
```
<site name="site_object" pos="-0.2 0 0.05" size="0.05" type="sphere" rgba="1 1 0 1"/>
<geom name="geom_object" pos="0.2 0 0.05" size="0.05" type="sphere" rgba="1 1 0 1"/>
```
code:
```
model.site('site_object').pos = [0.4, 0, 0.05]
model.geom('site_object').pos = [-0.4, 0, 0.05]
mujoco.mj_step(model, data)
viewer.sync()
```
</details>
But, a site does not have collision and a simple geom is fixed to the world body.

When I put the geom inside of a body to give it a free joint, like so:
 <details>
  <summary>geom object inside a body in world body</summary>

mjcf:
```
<body name="body_object" pos="0 0.2 0.05">
      <freejoint/>
      <geom name="body_object_geom" pos="0.2 0 0.05" size="0.05" type="sphere" rgba="1 1 0 1"/>
</body>
```
</details>
I'm unable to move even the geom inside in the way it was done before. I've tried changing the `pos` value for both the geom and body through the `model` and even tried changing the `xpos` value of both through `data`, but nothing works. I assume there's a way to do this since it's common to simulate simple objects like this, but how do I do so in mujoco?

yuvaltassa, 2023-09-16:
You gave it a freejoint, just set the qpos value of the freejoint?

manavkulshrestha, 2023-09-18:
Hi! how should I access the joint's qpos? neither `model.body('body_object').qpos` nor `data.body('body_object').qpos` seem to work

yuvaltassa, 2023-09-18:
qpos is associated with the joint, not the body.

manavkulshrestha, 2023-09-18:
1. Is there some way to access the joints associated with a certain body and then index into the qpos?
2. once the qpos is modified and qvel is set to 0, why is it that object continues to move if it was previously bumped? Is there a force vector stored somewhere?

yuvaltassa, 2023-09-18:
1. Of course. <joint name="myjoint"/> and then data.joint("myjoint").qpos Try searcging in the docs before asking here. This is all [explained](https://mujoco.readthedocs.io/en/latest/python.html#named-access)

2. No idea what you mean. Add a MWE if you want us to diagnose.

yuvaltassa, 2024-01-02:
Closing due to lack of activity.

closed 2024-01-02 by yuvaltassa