google-deepmind/mujoco #1058: Simulate a flexible rod

fukangl, 2023-09-17:
Hi, I am new to Mujoco and am currently attempting to simulate a soft robot arm in Mujoco. Is there a reference or guide available that can help me construct a model of a flexible metal rod(as shown in the red color part of the picture) in Mujoco?

[image]

yuvaltassa, 2023-09-18:
https://www.youtube.com/watch?v=4DvGe-BodFU

https://github.com/google-deepmind/mujoco/blob/main/model/plugin/elasticity/coil.xml

closed 2023-09-18 by yuvaltassa

fukangl, 2023-09-26:
Hi @yuvaltassa , 

I employed four slider joints to manage the tendon's motion. The actuator settings are as follows.  However, there is an unexpected behavior where moving one slider in a particular direction causes the other sliders to move as well. Is there a solution to rectify this issue?

``` 
<actuator >
        <!--        -->
        <position name="slider_d11" joint="slider_d11" ctrllimited="false"  ctrlrange="-0.3 0.3"  kp="100"/>
        <position name="slider_d12" joint="slider_d12" ctrllimited="false"  ctrlrange="-0.3 0.3"  kp="100"/>
        <position name="slider_d21" joint="slider_d21"  ctrllimited="false" ctrlrange="-0.3 0.3"  kp="100"/>
        <position name="slider_d22" joint="slider_d22"  ctrllimited="false" ctrlrange="-0.3 0.3"  kp="100"/>
</actuator>
 ```

[image]

yuvaltassa, 2023-09-28:
First of all, this is a very cool model!

1. Regarding your message from 2 days ago, it's not clear to me what is the problem that you're trying to fix. Sounds like maybe you want higher gains in your actuators?

2. Regarding "Is there a method to enable the tendon to pass through multiple geoms without requiring the insertion of sites between them?", the answer is no. (This makes the tendon path not analytically computable).

3. "Alternatively, are there strategies to avoid the pulling forces between the disks in the simulation?" It's not clear to me what you are asking, how could the pulling forces be avoided, this seems to not make physical sense.

Here's what I propose. Delete all the geoms in the path of your tendons, use only sites. I think this will help you reason about your model, also it's not at all clear that you need the geoms at all as the holes are so small.

fukangl, 2023-09-28:
Hi @yuvaltassa , Thank you for your reply!

1.For the first question, our objective is to maintain the static position of other actuators while controlling the movement of a single actuator in one direction. For instance, when we command one actuator to move 10cm in a specific direction (as depicted in the image on the right below), it is essential that the remaining actuators remain stationary (see the image on the left below).

[image]

2. If we route the tendon through multiple sites on the disks, it will cause these disks to be pulled, which differs from real-world behavior. In reality, the tendon should only exert pulling force on the disk to which one of its ends is connected. The interaction between the tendon and other disks is primarily characterized by the pressure and friction applied to the holes.

yuvaltassa, 2023-09-28:
1. I recommend tweaking actuator gains and possibly adding damping (in case high gains lead to instability).
2. The tendons are not "tied" to sites on the path, they are only **way points**. A site on the path is like a hole of 0 radius, so I'm not sure I understand what you're saying.

fukangl, 2023-09-28:
So, What kinds of forces does the tendon exert on the disks with holes?

yuvaltassa, 2023-09-28:
It can only push on it sideways (no friction). More generally, think about the change in `qpos` that would cause a change to the **length of the tendon**. That (and only that) is the direction of force that the tendon can apply.

fukangl, 2023-10-06:
We choose to affix one end of the tendon to a specific site within a world body. To actuate the tendon, we reset the position of the world body. Our objective is to maintain a constant length for the tendon throughout this operation. However, an issue arises as the length of the tendon falls outside the designated range (e.g., range="0.455 0.47"), with the sensor monitoring value measuring approximately 0.42. Are there any solutions to address this problem?

fukangl, 2023-10-10:
When we employ equality constraints to regulate the length of the tendon, it appears that the tendon's length still exceeds the specified range. Could you please provide some guidance on how to address this issue?

yuvaltassa, 2023-10-10:
Constraints in MuJoCo are not hard constraints. Use the [appropriate attributes](https://mujoco.readthedocs.io/en/latest/modeling.html#solver-parameters) to make them as hard as you like (note that these apply both to the bilateral (equality) and unilateral (spatial) solver parameters). Probably the easiest thing is to make the range a bit smaller. I'm sure your physical tendon can also stretch a bit. You can try to do some careful system identification to get constraint parameters that match your tendon material.

yuvaltassa, 2023-10-10:
Surely there is a force on the tendon that is fighting with the constraint?

Make a model that only has the tendon and play with the solver parameters of whatever constraint you are using to understand the relationship. You basically have a nonlinear elastic force-length relationship that you can tune.

jskiit, 2025-06-18:
Is it possible to send the code for this model, I am trying to model as similar one, but i am getting errors like the model is oscillating up and down, the code will be helpful to debug my code