google-deepmind/mujoco #1059: "Import mujoco" crashes on Macbook M2

Drxlalala, 2023-09-18:
**How to ask for help**

First, read our quick guide to
[asking good questions](https://github.com/google-deepmind/mujoco#asking-questions).
Below is a template for you to use:

Hi,

I'm a student  and I'm trying to use MuJoCo for training UR5 to learn SAC.

I'm looking for some help with install MuJoCo.

Here is a model which explains my question:

When I installed mujoco210 on Macbook M2, I write the code in Terminal with "./simulate ../model/humanoid.xml " 
It will show this erro " illegal hardware instruction "
How can I deal with problem?
Thanks a lot 
<details>
  <summary>minimal XML</summary>

```XML
<mujoco>
  <worldbody>
    <light pos=".4 -.4 .3" dir="-2 2 -1.5" diffuse=".6 .6 .6"/>
    <light pos="-.2 -.4 .3" dir="1 2 -1.5" diffuse=".6 .6 .6"/>
    <geom type="plane" size=".5 .5 .01"/>
    <body name="ball" pos="0 0 0.05">
      <freejoint/>
      <geom size=".03"/>
    </body>
    <body name="hammer" pos="-.05 0 0.18">
      <joint axis="0 1 0"/>
      <geom type="capsule" size=".01" fromto="0 0 0 -.15 0 0"/>
      <geom type="box" size=".02 .02 .03" pos="-.15 0 0"/>
    </body>
  </worldbody>
</mujoco>
```

</details>

Here is a screenshot , illustrating my question:
[image]

saran-t, 2023-09-19:
MuJoCo 2.1.0 doesn't support Apple Silicon. Why are you using such an old version?

Drxlalala, 2023-09-19:
> MuJoCo 2.1.0 不支持 Apple Silicon。为什么要用这么旧的版本？

Bacause I need to import mujoco_model in python and train the model. So I need to use mujoco_py, and the last mujoco_py is to be compatible with MuJoCo version 2.1

Drxlalala, 2023-09-19:
> MuJoCo 2.1.0 不支持 Apple Silicon。为什么使用这么旧的版本？

closed 2023-09-19 by Drxlalala

saran-t, 2023-09-19:
You could try version 2.1.1 and see if it works anyway.