google-deepmind/mujoco #1061: What is the best way to add camera sensors to mujoco and render that data?

alishbaimran, 2023-09-20:
Hi,

I'm a student  and I'm trying to use MuJoCo for a research project.

I'm looking for some help with two cameras as the eyes for a humanoid robot. I would like for these cameras to move as the body moves. As well, I'd like to render this data. What is the best way to do this?

yuvaltassa, 2023-09-20:
Put the camera on bodies with joints, and then render from those cameras as required.

closed 2023-09-20 by yuvaltassa

alishbaimran, 2023-09-20:
Could you provide an example of the code for what joints to add, where and where to add the cameras? As mentioned, i want to add two cameras (one for each eye on the head). So the camera view should be from the POV of the robot and move with the body. For context, the head of my robot is defined within <worldbody> as a <body>:

<worldbody>
....
      <body name='head' pos='0.02 0 0.62' euler='1.5708 1.5708 0'>
        <geom type='mesh' mesh='head' material='head'/>
        <geom class="collision" type="mesh" mesh="head" friction='0.7 0.01 0.005'/>
        <joint name='head-roll' type='hinge' axis='1 0 0' range='-0.523599 0.523599' stiffness='0' frictionloss='1' damping='1' armature='0.0001' limited='true' pos='0 0 0'/> 
        <joint name='head-pitch' type='hinge' axis='0 1 0' range='-0.523599 0.523599' stiffness='0' frictionloss='1' damping='1' armature='0.0001' limited='true' pos='0 0 0'/>
        <joint name='head-yaw' type='hinge' axis='0 0 1' range='-0.523599 0.523599' stiffness='0' frictionloss='1' damping='1' armature='0.0001' limited='true' pos='0 0 0'/>
      </body>
....
</worldbody>

alishbaimran, 2023-09-20:
I believe the issue is with the positioning of my cameras. How do I figure out where the cameras are and how to fix the positioning?

yuvaltassa, 2023-09-20:
simulate has camera visualisation, and you can also look through the cameras. There is also a "copy camera" button that copies into the clipboard an XML snippet of the free camera as a fixed camera