google-deepmind/mujoco #1062: How can I generate a soft contact?

Uptomylimit, 2023-09-20:
Hi,

I'm a student and I'm trying to use MuJoCo for simulation.

I'm looking for some help with setting contact parameters.

How can I generate a soft contact? So when an object contact another soft object, it will not generate a big force in normal direction and make the object fly away?

Balint-H, 2023-09-21:
Check out the `solref` and `solimp` parameters of the colliding geoms (https://mujoco.readthedocs.io/en/stable/modeling.html#csolver). Check out some of the options available for defining geom interaction (e.g. margin) https://mujoco.readthedocs.io/en/stable/XMLreference.html#body-geom-solref

You can set these values in your xml of your geom.

Uptomylimit, 2023-09-22:
> Check out the `solref` and `solimp` parameters of the colliding geoms (https://mujoco.readthedocs.io/en/stable/modeling.html#csolver). For example you can set a larger width in `solimp`, or a larger time constant in `solref`, depending on the exact behaviour you are looking for. You could also use the stiffness-damping format in `solref` (by using negative values), which might make it easier to interpret (and use a low stiffness value).
> 
> Check out some of the options available for defining geom interaction (e.g. margin) https://mujoco.readthedocs.io/en/stable/XMLreference.html#body-geom-solref
> 
> You can set these values in your xml of your geom.

Thanks for your reply!

I spend a lot of time trying different solimp and solref parameters and find that the contact is hard to set.  Finally I find some parameters that is good. And thanks for your reply again!

closed 2023-09-27 by saran-t

Josh00-Lu, 2024-07-16:
> Thanks for your reply!
>
> I spend a lot of time trying different solimp and solref parameters and find that the contact is hard to set. Finally I find some parameters that is good. And thanks for your [...]

Hello, can you explain what parameters are good? Same problem, thanks so much.