google-deepmind/mujoco #1066: Using `sys/stat stat()` for checking file existence leads to sigfault error of `mjv_makeScene`

someilay, 2023-09-22:
Hi! During use of `mujoco` I faced the following bug (at least I think so):

Setup:
`ubuntu22.04, kernel: 6.2.0-31-generic`
`mujoco v2.3.7`
`glfw v3.3.8`
`cmake v3.22.1`
`gcc v11.4.0`
`c++ 17, `

Code snippet:
```c++
#include <sys/stat.h>

bool exists(const std::string &name) {
    struct stat buffer{};
    int fd = stat(name.c_str(), &buffer);
    if (fd > -1) close(fd);
    return fd == 0;
}
...
char error[1000];
exists("main.cpp"); // comment this line removes problem with mjv_makeScene
mjModel* m = mj_loadXML("../example.xml", nullptr, error, 1000);
mjData* d = mj_makeData(m);
mjvCamera cam;
mjvOption opt;
mjvScene scn;
mjrContext con;
// init GLFW, create window, make OpenGL context current, request v-sync
glfwInit();
GLFWwindow* window = glfwCreateWindow(1200, 900, "Demo", nullptr, nullptr);
glfwMakeContextCurrent(window);
glfwSwapInterval(1);

// initialize visualization data structures
mjv_defaultCamera(&cam);
mjv_defaultOption(&opt);
mjr_defaultContext(&con);

// create scene and context
mjv_makeScene(m, &scn, 1000);  // <- sigfault appears here
```

The `exists` function works fine and doesn't contain any errors by itself. It's a slightly changed version from [Stackoverflow](https://stackoverflow.com/questions/12774207/fastest-way-to-check-if-a-file-exists-using-standard-c-c11-14-17-c)

If I remove (comment call) this function, the above code works fine. Moreover, if `exists` would be called after `mjv_makeScene`, it would also work without any error.

The overall code is taken from the official [documentation](https://mujoco.readthedocs.io/en/latest/programming/visualization.html). I only added a model loading. The `.xml` is taken from [here](https://mujoco.readthedocs.io/en/latest/_static/example.xml).

I compile this code using the following cmake file:
```cmake
cmake_minimum_required(VERSION 3.22)
project(example VERSION 1.0)

if(NOT CMAKE_BUILD_TYPE)
    message(STATUS "Set Release as default")
    set(CMAKE_BUILD_TYPE Release)
endif()

if(NOT CMAKE_CXX_EXTENSIONS)
    message(STATUS "Disabling CMAke extensions")
    set(CMAKE_CXX_EXTENSIONS OFF)
endif()

if (NOT MUJOCO_DIR)
    message(FATAL_ERROR "Specify MUJOCO_DIR!")
endif()

if (NOT GLFW_DIR)
    message(FATAL_ERROR "Specify GLFW_DIR!")
endif()

# Connect GLFW
find_package(glfw3 3.3 REQUIRED PATHS ${GLFW_DIR})
find_package(OpenGL REQUIRED)

# Set standards
set(CMAKE_CXX_STANDARD 17)
set(CMAKE_CXX_FLAGS "-Wextra")
set(CMAKE_CXX_FLAGS_DEBUG "-g")
set(CMAKE_CXX_FLAGS_RELEASE "-O3")

# add include directories
include_directories(${MUJOCO_DIR}/include include)
link_directories(${MUJOCO_DIR}/lib)

# add the source directories
# file(GLOB SRCS src/*.cpp)

add_executable(example main.cpp)
target_link_libraries(example glfw OpenGL::GL mujoco)
```

Where `MUJOCO_DIR` is compiled binaries taken from `GitHub`; `GLFW_DIR` is compiled and installed binaries of `glfw`

The command that I use: `cmake -S . -B ./build -DMUJOCO_DIR=../mujoco-2.3.7 -DGLFW_DIR=../glfw-install`

saran-t, 2023-09-22:
I haven't tried to build and repro this issue, but am pretty certain that you're just missing a call to `mjv_defaultScene` before your `mjv_makeScene`.

someilay, 2023-09-22:
@saran-t. Thanks for your response. It helped in my case. Anyway, I'm still wondering why my code worked without `mjv_defaultScene`. This function wasn't mentioned in the [docs](https://mujoco.readthedocs.io/en/latest/programming/visualization.html).

closed 2023-09-22 by someilay

yuvaltassa, 2023-09-22:
It's in the [API reference](https://mujoco.readthedocs.io/en/stable/APIreference/APIfunctions.html#mjv-defaultscene), but I agree that perhaps it should be mentioned in the section you liked to. Feel free to send a PR to improve the docs.

saran-t, 2023-09-22:
This is a mistake in the doc. The problem is that `mjv_makeScene` [unconditionally calls `mjv_freeScene`](https://github.com/google-deepmind/mujoco/blob/main/src/engine/engine_vis_init.c#L122), which in turn unconditionally calls `free` on a bunch of pointers.

Without `mjv_defaultScene` the `scn` variable is uninitialised memory. Arguably the `make` family of functions shouldn't be in the business of `free`ing the previous copy at all. @yuvaltassa WDYT of changing the semantics of these functions? Too risky for existing code?

yuvaltassa, 2023-09-22:
@saran-t that's a good point. Though not having that free would be a memory leak footgun, no?

I think better documentation is all we need here.

saran-t, 2023-09-22:
It's pretty unusual for malloc-style function to free things, I'd argue that this is unexpected behaviour. Yes there's leak risk, but segfault risk for calling a "constructor" on uninitialised memory is an even bigger footgun IMO. Constructors are reasonably expected to correctly initialise structs in any configuration (and the risk of memory leak when calling constructors on already-initialised structs is usually expected).

yuvaltassa, 2023-09-22:
Seems to me that both the current behaviour and what you're proposing requires clear documentation to prevent user error. In that case I prefer improving documentation for the current behaviour, rather than changing the behaviour. Does @someilay or perhaps @nimrod-gileadi have an opinion?

saran-t, 2023-09-22:
Another way to proceed is to add a guard field into these structs, and have `make` only call free if the field is set to the correct magic value.

someilay, 2023-09-22:
@yuvaltassa. The proposed solution to use `mjv_defaultScene` has worked in my case. However, I can get how the initialization of the scene is connected with a `stat` function. Without any initialization and calling `exists` it works well. However, checking the file's existence force to use  `mjv_defaultScene` .  For me, definitely, somewhere in the `mujoco` code exists mentioned `unexpected behavior`.