google-deepmind/mujoco #1072: What is the frame of `qvel` of a floating body?

JewelryForge, 2023-09-27:
I'm trying to figure out what the frame of `qvel` of a floating body is. The doc says it is "3D linear velocity followed by 3D angular velocity" but does not specify its frame. It's clear that the angular velocity is in the body frame, but the linear velocity seems to be in neither of the two frames.

v-r-a, 2023-09-27:
Explanation provided in issue #691 may help you.

yuvaltassa, 2023-09-27:
Closing the issue as @v-r-a has answered, but I feel like since this has come up more than once, the documentation should be updated to explain this more clearly. Where do you think is the appropriate place in the documentation to add this explanation?

closed 2023-09-27 by yuvaltassa

JewelryForge, 2023-09-27:
> Perhaps [this section](https://mujoco.readthedocs.io/en/latest/overview.html#floating-objects) should be expanded/improved?

I guess so.