google-deepmind/mujoco #1074: Position and torque control switch during the simulation

QiyangYan, 2023-10-02:
Hi,

I'm a student and I'm trying to use MuJoCo for a research project.

I'm looking for some help with **actuator-type switching during the simulation**.

My robot requires the actuator to switch between position control and torque control during the task, but I could find this type of actuator or relevant info on the website regarding the type switch, I'm wondering if I missed anything and if you have any advice on how should I approach this?

nimrod-gileadi, 2023-10-02:
It would be a bit hard to encapsulate that within a single XML MuJoCo model.

I would say the cleanest option is to use torque actuators, and then implement a PD control law yourself in the client code which uses MuJoCo, to convert from goal positions to torques.

Alternatively, you could look at the documentation for `<motor>`, `<position>` and `<velocity>` ([link](https://mujoco.readthedocs.io/en/stable/XMLreference.html#actuator-motor)), and figure out how you need to update `model.actuator_gainprm`, `model.actuator_biastype` and `model.actuator_biasprm` to go from a torque actuator to a position actuator. You probably would want to change `actuator_ctrlrange`.

QiyangYan, 2023-10-02:
Thanks for your tips! I will look into these two methods!

QiyangYan, 2023-10-02:
Hi, the second method works, but there is one more thing I would like to confirm.

Comparing the position and torque actuator shortcut, are these three parameters you mentioned and the user-defined params like actuator_ctrlrange the only difference they have?

kevinzakka, 2023-10-02:
Regarding @nimrod-gileadi's second proposal, check out this neat [Jupyter notebook](https://github.com/lvjonok/mujoco-actuators-types) from @lvjonok that changes the actuator types at runtime.

closed 2023-10-02 by kevinzakka