google-deepmind/mujoco #1075: Set actuator parameters for Dynamixel xm430 in MuJoCo

QiyangYan, 2023-10-02:
Hi,

I'm a student and I'm trying to use MuJoCo for Sim2Real of a robot with [Dynamixel motors XM430](https://emanual.robotis.com/docs/en/dxl/x/xm430-w350/#specifications).

I'm looking for some help with parameter setting for the actuator.

My question is:
How should I configure the parameters for a position-controlled motor to minimize the discrepancy between its simulated positional offset and the actual offset caused by external forces? While I can decrease the offset by using a high damping value and a large gear ratio, is there a more systematic or recommended approach to determine these values based on the motor's specifications?

saran-t, 2023-10-02:
Those are the servos used in the OP3 robot, which we've successfully done SysID and sim2real for. Check out the [OP3 model](https://github.com/google-deepmind/mujoco_menagerie/tree/main/robotis_op3) from Menagerie.

QiyangYan, 2023-10-02:
Oh! Great! This makes the work much easier! Apart from the xm430, our robot also uses a smaller Dynamixel xl320 servo, which moves a geom in a trajectory, that's similar to a slide joint, to lift objects. To simply the mechanism for simulation, I've replaced this with a slide joint. Do you have any recommendations, guidelines, or resources on how to configure the position controller for this slide joint?

yuvaltassa, 2023-10-10:
Well you really want to do proper system ID, like we did for the original motors. We intend to publish a set of guidelines and best practices for sysID later this year. TL;DR

$\arg\min_\theta(||\textrm{simulated}(\theta)-\textrm{measured}||^2)$

closed 2023-10-10 by yuvaltassa

QiyangYan, 2024-02-03:
Hi, I'm wondering if the guideline is released?

yuvaltassa, 2024-02-05:
> Hi, I'm wondering if the guideline is released?

Not yet. Will update here when it is.

mizrex, 2024-03-21:
> Those are the servos used in the OP3 robot, which we've successfully done [sim2real transfer](https://www.youtube.com/@OP3Soccer) on. Check out the [SysID'd OP3 model](https://github.com/google-deepmind/mujoco_menagerie/tree/main/robotis_op3) from Menagerie.

I'm looking into system identification. The [paper ](https://arxiv.org/abs/2304.13653)mentioned your team used a sinusoidal control signal with varying frequency on a motor with a known load attached to the motor. If possible, can you help with the following questions:
1. How did you define the frequency range of the sinusoidal control signal. I tried using minimize functions (eg. least squares) to match the motor response. But I think knowing the frequency range of the varying control signal would be critical for optimal system identification.
2. Did you do system identification on one motor independently, or the whole system together? If it was done for the whole system, how did you manage to run tests on the biped (when it's not balancing), as gravity will act differently based on how it's standing.
Any guidance would be helpful.

yuvaltassa, 2024-03-21:
1. I shouldn't matter much. Just use a sensible chirp?
2. One motor.

mizrex, 2024-04-02:
> Those are the servos used in the OP3 robot, which we've successfully done [sim2real transfer](https://www.youtube.com/@OP3Soccer) on. Check out the [SysID'd OP3 model](https://github.com/google-deepmind/mujoco_menagerie/tree/main/robotis_op3) from Menagerie.

The paper mentions you got better results with sim2real transfer using position control mode actuators, vs direct current control actuators. Did direct current control actuators work at all? (we are currently defining direct current control actuators in our xml)
We did system Id on a single motor and trained policies around them. However, running into issues with making the real biped walk (it works in the simulation). I'm trying to find out the discrepancies causing it.

If I were to do a system Id on the complete system (using all the 10 motors). What would be the best way to go about doing it? Previously for a single motor, we had tried using functions like curve fitting on a single motor dataset. 

Thank you for all your help.

kevinzakka, 2026-02-11:
👀