google-deepmind/mujoco #1079: Mujoco installtion failed

raviwaarr, 2023-10-04:
Hi,

I'm a student and I'm trying to use MuJoCo for RL research.

I'm looking for some help with installing mujoco on my apple sillicon m1. It's stuck on MUJOCO PATH.

Here is a model which explains my question:

 Building wheels for collected packages: mujoco
  Building wheel for mujoco (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [68 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.macosx-10.9-x86_64-cpython-311
      creating build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/viewer.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/renderer.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/renderer_test.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/rollout.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/rollout_test.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/bindings_test.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/viewer_test.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/render_test.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/gl_context.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      creating build/lib.macosx-10.9-x86_64-cpython-311/mujoco/cgl
      copying mujoco/cgl/cgl.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco/cgl
      copying mujoco/cgl/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco/cgl
      creating build/lib.macosx-10.9-x86_64-cpython-311/mujoco/egl
      copying mujoco/egl/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco/egl
      copying mujoco/egl/egl_ext.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco/egl
      creating build/lib.macosx-10.9-x86_64-cpython-311/mujoco/osmesa
      copying mujoco/osmesa/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco/osmesa
      creating build/lib.macosx-10.9-x86_64-cpython-311/mujoco/glfw
      copying mujoco/glfw/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco/glfw
      running build_ext
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/private/var/folders/25/3c_cvnxx415_zr7xxz5k1_zh0000gn/T/pip-install-dgv42c3q/mujoco_7288605b277e43ba90de2e1b28d696dc/setup.py", line 418, in <module>
          setup(**SETUP_KWARGS)
        File "/Users/raviwar/miniconda/lib/python3.11/site-packages/setuptools/__init__.py", line 103, in setup
          return distutils.core.setup(**attrs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/Users/raviwar/miniconda/lib/python3.11/site-packages/setuptools/_distutils/core.py", line 185, in setup
          return run_commands(dist)
                 ^^^^^^^^^^^^^^^^^^
        File "/Users/raviwar/miniconda/lib/python3.11/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
          dist.run_commands()
        File "/Users/raviwar/miniconda/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
          self.run_command(cmd)
        File "/Users/raviwar/miniconda/lib/python3.11/site-packages/setuptools/dist.py", line 989, in run_command
          super().run_command(command)
        File "/Users/raviwar/miniconda/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/Users/raviwar/miniconda/lib/python3.11/site-packages/wheel/bdist_wheel.py", line 364, in run
          self.run_command("build")
        File "/Users/raviwar/miniconda/lib/python3.11/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "/Users/raviwar/miniconda/lib/python3.11/site-packages/setuptools/dist.py", line 989, in run_command
          super().run_command(command)
        File "/Users/raviwar/miniconda/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/Users/raviwar/miniconda/lib/python3.11/site-packages/setuptools/_distutils/command/build.py", line 131, in run
          self.run_command(cmd_name)
        File "/Users/raviwar/miniconda/lib/python3.11/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "/Users/raviwar/miniconda/lib/python3.11/site-packages/setuptools/dist.py", line 989, in run_command
          super().run_command(command)
        File "/Users/raviwar/miniconda/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/private/var/folders/25/3c_cvnxx415_zr7xxz5k1_zh0000gn/T/pip-install-dgv42c3q/mujoco_7288605b277e43ba90de2e1b28d696dc/setup.py", line 156, in run
          self._mujoco_framework_path) = self._find_mujoco()
                                         ^^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/25/3c_cvnxx415_zr7xxz5k1_zh0000gn/T/pip-install-dgv42c3q/mujoco_7288605b277e43ba90de2e1b28d696dc/setup.py", line 170, in _find_mujoco
          raise RuntimeError(
      RuntimeError: MUJOCO_PATH environment variable is not set
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for mujoco
  Running setup.py clean for mujoco
Failed to build mujoco

aravindsiv, 2023-10-05:
From [here](https://github.com/google-deepmind/mujoco/blob/main/doc/python.rst#building-from-source): you need to set the `MUJOCO_PATH` environment variable (e.g., `export MUJOCO_PATH=/Applications/MuJoCo.app`).

yuvaltassa, 2023-10-05:
Thanks @aravindsiv, note I edited your comment to point to the proper docs server. GitHub RST rendering is pretty janky.

closed 2023-10-05 by yuvaltassa

saran-t, 2023-10-05:
BTW, @raviwaarr unless you are modifying the Python bindings you should prefer `pip install mujoco`.

If you need to install MuJoCo at GitHub HEAD (i.e. newer than latest release), try `pip install mujoco --pre -f https://py.mujoco.org/`.