google-deepmind/mujoco #1080: Recompute inertia on the fly

vikashplus, 2023-10-04:
I have a body with a few geoms. 

```xml
<body name="Object" pos="0 0 1">
    <geom name="geom1" type="box" size="0.02 0.02 0.02"/>
    <geom name="geom2" type="box" size="0.02 0.02 0.02"/>
    <geom name="geom3" type="box" size="0.02 0.02 0.02"/>
</body>
```

I'm trying to programmatically morph this body into random arbitrary objects by randomizing the `geom_types` and `geom_size`, `geom_pos`, `geom_quat`. I am also adjusting the `geom_rbound` and `geom_aabb` accordingly for proper collisions.

Is there a way to recompute inertial properties of this body after the shapes has been regenerated?

yuvaltassa, 2023-10-06:
Is pyMJCF not a good solution?

vikashplus, 2023-10-06:
I want to be able to do it programmatically post-load in the compiled model. My understanding is that pyMJCF is only for XML generation. Let me know if I am missing anything.

vikashplus, 2023-10-07:
For my use case - 
- I need to be able to do it at a really high speed. 
- The models are huge and load time is significant.
- The sim instance can't be destroyed and recreated

yuvaltassa, 2023-10-08:
In that case you need to re implement the inertia-inference logic in the compiler. It's not very complicated.
Let me know if you need help finding the code or have further questions.

closed 2023-10-08 by yuvaltassa

vikashplus, 2023-10-09:
It would be helpful if I could get a few pointers and I'll take it from there.

yuvaltassa, 2023-10-16:
Start [here](https://github.com/google-deepmind/mujoco/blob/110ade14356cb99c5aa82e479998fcd2a9d8c4ed/src/user/user_objects.cc#L1386)