google-deepmind/mujoco #1082: Dynamic scene update in passive viewer

guyazran, 2023-10-04:
Hi,

I'm an engineer and I'm trying to use MuJoCo for building environments for robotic navigation tasks. I'm looking for some help with adding markers to the scene to indicate the goal location in the task when using the built-in mujoco passive viewer, that is, `viewer = mujoco.viewer.launch_passive(model, data)`. Is it possible to add arbitrary decoration geoms to the scene and have them appear in the viewer without adding them to the model?

consider this simple model of a sphere floating in 3d space:

<details>
  <summary>Floating sphere example</summary>

```XML
<mujoco model="floatworld">
    <!-- objects are floating around in space-->
    <option gravity="0 0 0" />

    <asset>
        <!-- bright skybox background for rendering visibility-->
        <texture type="skybox" builtin="gradient" rgb1="1 1 1" rgb2=".6 .8 1" width="256"
            height="256" />
    </asset>
    <worldbody>
        <!-- lighting for visibility-->
        <light diffuse=".5 .5 .5" pos="0 0 3" dir="0 0 -1" />

        <body name="ball">
            <!-- three DOF for translation in all directions -->
            <joint name="slide_x" type="slide" pos="0 0 0" axis="1 0 0" damping="1"/>
            <joint name="slide_y" type="slide" pos="0 0 0" axis="0 1 0" damping="1"/>
            <joint name="slide_z" type="slide" pos="0 0 0" axis="0 0 1" damping="1"/>
            <geom type="sphere" pos="0 0 0" size=".1 .1 .1" rgba="1 0 0 1"/>
        </body>

    </worldbody>

    <actuator>
        <!-- actuation for ball DOFs-->
        <motor name="motor_x" joint="slide_x" gear="1" ctrlrange="-10 10" ctrllimited="true"/>
        <motor name="motor_y" joint="slide_y" gear="1" ctrlrange="-10 10" ctrllimited="true"/>
        <motor name="motor_z" joint="slide_z" gear="1" ctrlrange="-10 10" ctrllimited="true"/>
    </actuator>
</mujoco>

```

</details>

Assume that the goal is for the sphere to navigate to several random locations on the unit sphere. since the number of locations is not known in advance, we cannot add them to the xml model. It _is_ possible to set a maximum number of goal locations, but this (a) limits our environment, and (b) clutters up the XML.

I tried to emulate the behavior of the [_add_marker_to_scene](https://github.com/Farama-Foundation/Gymnasium/blob/54d2ba79404e9937ba2a1d44449915ecb00f95ce/gymnasium/envs/mujoco/mujoco_rendering.py#L81) in gymnasium. See [my screenshot](https://github.com/google-deepmind/mujoco/assets/12692763/177091c9-d844-4b68-baae-21366b206cc0) on how this is able to achieve the desired result with a single goal location. This solution updates the `MjvScene` struct by adding more geoms to the scene after updating it with the current simulation state. Trying this with the passive viewer, the scene object is immediately reverted once the `viewer.lock()` is released, even before the call to `viewer.sync()`. Is there a way to overcome this?

yuvaltassa, 2023-10-05:
This is WIP, updates soon.

saran-t, 2023-10-10:
We're planning to release a new version that incorporates 2e15574b58df4bfda59e1642cf67699cb0a58667 in the middle of next week.

If you want to try this out before then, you can pip install mujoco --pre -f https://py.mujoco.org/.

closed 2023-10-10 by saran-t