google-deepmind/mujoco #1083: Contact force has a sudden jump in some time steps

Haichao-Zhang, 2023-10-05:
Hi,

I'm a  researcher and I'm trying to use MuJoCo for grasping related task.

I'm looking for some help with contact force.

I used ``[mj_contactForce](https://mujoco.readthedocs.io/en/latest/APIreference/APIfunctions.html#mj-contactforce)`` to extract the contact force (between gripper finger and object to be grasped) and inspected its first component (force along normal direction). 

After been grasped and lifted, most of the time, the force looks normal, but occasionally, some large magnitude normal forces appears, e.g.:

```
t_k: 0.45, t_{k+1}: 0.42, t_{k+2}: -81.0 ....
```

Looking for your help on understanding why this happens and suggestions on how to fix it.

Thanks!

kevinzakka, 2023-10-05:
Are you using dm_control Physics by any chance?

kevinzakka, 2023-10-05:
See this [previous issue](https://github.com/google-deepmind/mujoco/issues/13) for an explanation. One solution would be to toggle the recently added `legacy_step=False` attribute of your Physics object.

closed 2023-10-05 by kevinzakka

Haichao-Zhang, 2023-10-06:
> See issue #13 for an explanation. One solution would be to toggle the recently added `legacy_step=False` attribute of your Physics object.

Thanks for the quick reply! Confirmed that this is indeed the reason. Thanks for your help!