google-deepmind/mujoco #1084: How to calculate the joint friction force?

Uptomylimit, 2023-10-06:
Hi,

I'm a student and I'm trying to use MuJoCo for simulation.

I am confused about the attribute 'frictionloss' in body/joint. In your document, the 'frictionloss' attribute affects the dry friction(or static friction). So when the joint moves with a speed qv, can you tell me how mujoco calculate the Coulomb friction of the joint?

brownbrowny, 2023-10-06:
Hi I have a similar question. I wanted to convert a URDF robot model to a Mujoco model. Later on the goal is to use that model in Unity for applying RL with MLAgents. I followed the instructions of the Menagerie models but can't figure out how to properly apply gear transmissions, friction forces, torque limits, etc. At the moment my robot can be controlled with high unrealistic speed. I also have another file from the manufacturer but I don't now how to transfer the values within that into Mujoco XML properties. I am fairly new to robotics and the documentation didn't help me. For example how to come up with this part is unclear to me:
`<general gaintype="fixed" biastype="affine" gainprm="2000" biasprm="0 -2000 -200"/>`

Here is the manufacturer XML file which should contain all the necessary values
[myrobot_6dof.txt](https://github.com/google-deepmind/mujoco/files/12828984/myrobot_6dof.txt)

Thanks in advance. Please let me know if I should open a new issue for this question.

yuvaltassa, 2023-10-06:
@Uptomylimit, this is not Colomb friction, there is no cone. See [here](https://mujoco.readthedocs.io/en/2.3.7/computation.html#friction-loss).

@brownbrowny your question is completely different. You need to do system identification on your robot.

closed 2023-10-06 by yuvaltassa

Uptomylimit, 2023-10-06:
> @Uptomylimit, this is not Colomb friction, there is no cone. See [here](https://mujoco.readthedocs.io/en/2.3.7/computation.html#friction-loss).
> 
> @brownbrowny your question is completely different. You need to do system identification on your robot.

I know that the fricitonloss affect the static friction, which is an upper limit on the absolute value of the force that friction can generate. But I am confused that how mujoco compute the friction generated when the joint moves. How do you model this part?

yuvaltassa, 2023-10-06:
Explained [here](https://mujoco.readthedocs.io/en/2.3.7/computation.html#dual-problem)