google-deepmind/mujoco #1085: Choice of Integrators

Haichao-Zhang, 2023-10-06:
Hi,

I'm a researcher and I'm trying to use MuJoCo for RL-related tasks.

I'm looking for some suggestions on the choice of integrators.
My question is for grasping-like contact rich scenarios, which integrator is the most appropriate, among available choices Euler, RK4, implicit etc, in terms of anti-slipping and less penetrations?

I searched over existing issues and found some relevant ones : https://github.com/google-deepmind/mujoco/issues/548#issuecomment-1293417831
but are not fully aligned with my targeted scenario.

Thanks for your help!

yuvaltassa, 2023-10-06:
Integrators won't usually make a big difference for contacts. implicitfast is usually your best choice generally, RK4 and implicit are good in specific situations related to energy and momentum conservation. 

We should update the relevant section about tuning contacts. (And choosing integrators, I suppose)

Ping us here if it doesn't happen by the end of the month.

Haichao-Zhang, 2023-10-06:
Thanks for the reply and the general guidance. Looking forward to the updated section.

yuvaltassa, 2023-10-26:
See end of this section (in Latest docs branch)
https://mujoco.readthedocs.io/en/latest/computation/index.html#numerical-integration

Haichao-Zhang, 2023-10-26:
> See end of this section (in Latest docs branch) https://mujoco.readthedocs.io/en/latest/computation/index.html#numerical-integration

Thanks for the followup on this issue. Great update on the doc. Thanks!

commit 5231a335 (2023-10-25): Improve documentation regarding choice of integrator. Fixes #1085

PiperOrigin-RevId: 576546484
Change-Id: I7192a0c32e1aed8e201d1db953b657e605cd74f0