google-deepmind/mujoco #1086: Scaling down body size results in `ValueError: Error: mass and inertia of moving bodies must be larger than mjMINVAL`

Omer1Yuval1, 2023-10-08:
I have a MuJoCo xml model of a worm made of a series of box geometries, and an ellipsoid head with a free joint. This is meant to model C. elegans, which is a 1mm nematode.

When I scale down the model (in terms of geometry size and relative position), I get the following error in simulate.exe: ValueError: Error: mass and inertia of moving bodies must be larger than mjMINVAL.

To give a slightly broader context, I am training a PPO agent to make the worm model do sine wave undulations (as a simple test case). Hence, scaling down gradually was important to adjust parameters in order to maintain similar dynamics across scales. I started from a ~0.3m long worm model, and then scaled it down by factors of 0.5, 0.1, 0.05, 0.02 and 0.01.

Until 0.05, the model behaves similarly and training results are good. At 0.02, training curves (mean episode length and reward) look very different and unstable, and training stopped after 2M steps with the following error: mujoco.FatalError: HessianDirect: rank-defficient Hessian. At 0.01, training doesn't even start, and I get the following error: ValueError: Error: mass and inertia of moving bodies must be larger than mjMINVAL. In this last case, the xml model cannot even be loaded to simulate.exe (due to the same error).

I attached xml files for scales 1, 0.05, 0.02 and 0.01 ([worm.zip](https://github.com/google-deepmind/mujoco/files/12840170/worm.zip)).

I am using MuJoCo 2.3.7 on Windows 11.

yuvaltassa, 2023-10-08:
I hope you're not using MKS for this?

Switch to CGS
<https://mujoco.readthedocs.io/en/latest/overview.html#units-are-unspecified>
and your problems will go away.

Dr. Yuval Tassa | Senior Research Scientist | Google DeepMind |
***@***.***

Omer1Yuval1, 2023-10-09:
Thanks @yuvaltassa!

yuvaltassa, 2023-10-10:
The latest documentation has a full description of both [fluid force models](https://mujoco.readthedocs.io/en/latest/computation/fluid.html)

yuvaltassa, 2023-10-11:
I should also add that all of the numbers you specified above have units. Depending on what sort of scaling you are doing, you might need to change these, yes? [Here](https://github.com/google-deepmind/dm_control/blob/main/dm_control/locomotion/walkers/assets/fruitfly_v2/fruitfly.xml) is a model using CGS model with all these parameters.

Omer1Yuval1, 2024-04-04:
Many thanks @yuvaltassa!

closed 2025-07-24 by Omer1Yuval1