google-deepmind/mujoco #1089: MuJoCo.app (macOS) crashes after launch without anything

Nuofan-SUSTech, 2023-10-09:
**How to ask for help**

First, read our quick guide to
[asking good questions](https://github.com/google-deepmind/mujoco#asking-questions).
Below is a template for you to use:

Hi,

I'm a student and I'm trying to use MuJoCo for research.

I'm looking for some help with MuJoCo installation. 

I download `mujoco-2.3.7-macos-universal2.dmg` from release and followed [instructions](https://mujoco.readthedocs.io/en/latest/programming/#getting-started) to install, then I open MuJoCo.app but nothing shows like image below:

[image]

After I click the window, it crashes.

Here is the system information:
```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               simulate [12306]
Path:                  /Volumes/VOLUME/MuJoCo.app/Contents/MacOS/simulate
Identifier:            org.mujoco.mujoco
Version:               2.3.7 (2.3.7)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-10-10 00:41:16.4079 
OS Version:            macOS 14.0 (23A344)
Report Version:        12
Anonymous UUID:        8315BE8C-0BF9-5E93-3977-19B29D4623EC

Sleep/Wake UUID:       80B8618D-1AC6-439E-96F7-F967EA7A5EAA

Time Awake Since Boot: 100000 seconds
Time Since Wake:       8669 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000448
Exception Codes:       0x0000000000000001, 0x0000000000000448

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [12306]

VM Region Info: 0x448 is not in any region.  Bytes before following region: 4307090360
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      100b90000-100bdc000    [  304K] r-x/r-x SM=COW  ...acOS/simulate

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libmujoco.2.3.7.dylib         	       0x1016158d0 mjv_moveCamera + 208
1   AppKit                        	       0x194420c8c -[NSWindow(NSEventRouting) _reallySendEvent:isDelayedEvent:] + 652
2   AppKit                        	       0x19442082c -[NSWindow(NSEventRouting) sendEvent:] + 284
3   AppKit                        	       0x194ac7c08 -[NSApplication(NSEventRouting) sendEvent:] + 1556
4   simulate                      	       0x100b9a0d4 _glfwPlatformPollEvents + 128
5   simulate                      	       0x100ba8620 mujoco::Simulate::RenderLoop() + 19280
6   simulate                      	       0x100b9bdfc main + 640
7   dyld                          	       0x1906c5058 start + 2224
```

I don't know what is going on. Is there anything I did wrong? By the way, I also tried `mujoco-2.3.6-macos-universal2.dmg` and it also crashes.

saran-t, 2023-10-09:
Looks like there's a bug if you scroll your mouse wheel in the window without having a model loaded. We'll get this fixed, but in the meantime just refrain from scrolling the mouse wheel until you've loaded a model.

Nuofan-SUSTech, 2023-10-09:
> Looks like there's a bug if you scroll your mouse wheel in the window without having a model loaded. We'll get this fixed, but in the meantime just refrain from scrolling the mouse wheel until you've loaded a model.

Thanks so much! After drag a model into it, it works well.

commit a1b39d59 (2023-10-09): Fix a segfault in `simulate` on mouse scroll without a loaded model.

Fixes #1089

PiperOrigin-RevId: 571985561
Change-Id: I4acc7c2a6032d678023783646c3f00cfe30641f2
closed 2023-10-09 by Nuofan-SUSTech