google-deepmind/mujoco #1092: Toggle Actuators On/Off in simulate.cc

cstegel, 2023-10-11:
**Is your feature request related to a problem? Please describe.**

I have a scene where I control multiple slide and hinge joints using position actuators. To debug this scene, it would be helpful to adjust the controls in simulate.cc's UI and then disable one of the actuators so that force is no longer applied along one of the constraints temporarily. I am not aware of a way to do this in the current simulate.cc UI.

For example, imagine a crane with one actuator that pulls an object up/down and another that pushes it further/closer away. I want to see what happens if I pick the object up, move it further away, then disable the up/down force to see it drop to the ground. 

**Describe the solution you'd like**

I would like a button beside each of the control sliders to enable/disable the actuator:

[image]

(Add a button beside each of these sliders or make it click-able).

**Describe alternatives you've considered**

I know there's a "disable all actuators" button but this disables everything:

[image]

 I want to disable some actuators while keeping others active.

**Additional context**

I see the code for creating the control sliders is [here](https://github.com/google-deepmind/mujoco/blob/main/simulate/simulate.cc#L1032) but I can't figure out how to add something that would disable actuators.

yuvaltassa, 2023-10-11:
Duplicate of #1024

Unfortunately disabling actuators at runtime is currently not possible.

I have a draft feature that would add  `<option actuatorgroupdisable/>`, which uses actuator groups to choose which ones to disable. Defaults to `""` (all enabled) and then you could say `<option actuatorgroupdisable="1 3 4">` to mean disable all actuators in some groups.

In terms of UI, this could be tied to a new panel letting you enable/disable at runtime, or this could just use the current group diable panel (although we would need to change the default from only 0 1 2 enabled to all enabled), in this case actuator visualisation would be tied to actuator activation.

The main limitation of this approach would be that only 6 groups of actuators could be enable/disabled at runtime. WOuld that be good enough for you?

cstegel, 2023-10-11:
Hi Yuval,

Thank you for considering this and working on MuJoCo! 

Disabling actuator groups similar to disabling them being rendered works for my use case. I only need to disable a couple different actuators individually so 6 groups should be enough for me.

yuvaltassa, 2023-10-12:
I hope to do this next week. 
Please ping me back here if it hasn't happened by the end of October?

closed 2023-10-31 by copybara-service[bot]

yuvaltassa, 2023-11-01:
https://mujoco.readthedocs.io/en/latest/modeling.html#group-disable

commit 893c4042 (2023-10-31): Runtime disabling of actuators according to group.

Added `option-actuatorgroupdisable` attribute and associated `mjOption.disableactuator` integer bitfield, used to disable sets of actuators at runtime according to their group.

- The first 6 actuator groups are toggleable in the `simulate` viewer.
- Minor refactor and cleanup of actuator documentation.

https://youtu.be/H9qG9Zf2W44

Fixes #1092.

PiperOrigin-RevId: 578335600
Change-Id: I4cf663b90ea768e4380acfa2fe3b8c15c7cbb568