google-deepmind/mujoco #1093: multi-threading and data reproducibility

pmh5050, 2023-10-12:
Hello,

I'm a student, and I'm attempting to use MuJoCo for simulations.

I'm seeking assistance with multi-threading and ensuring data reproducibility.

According to the documents below, MuJoCo appears to be well-suited for parallel computing:
1) [Reproducibility](https://mujoco.readthedocs.io/en/latest/computation/index.html#reproducibility): `MuJoCo's simulation pipeline is entirely deterministic and reproducible.`
2) [Multi-threading](https://mujoco.readthedocs.io/en/latest/programming/simulation.html#multi-threading): `the unified data structure mjData which contains everything that changes over time.`

However, when I attempt to work with a thread pool, I encounter some issues related to reproducibility.

+ Each thread has its own mjData `data[i]`
+ Each thread performs the following command with every call:
```c++
mj_resetData(model, data[i]);
for(int i = 0; i < steps; i++)
{
    mj_step(model, data[i]);
}
```

It usually produces the same outcomes (e.g., `steps 151`, `steps 153`), but occasionally exhibits undesired behavior. I'm curious about what might be causing these different outcomes. Can you recommend a solution to address this issue?

`steps 151`
[image]

`steps 152` (see the lower-right image)
[image]

`steps 153`
[image]

aravindsiv, 2023-10-13:
From [here](https://mujoco.readthedocs.io/en/latest/computation/index.html#reproducibility): 

Have you tried disabling warmstarts like so:
```
<option>
  <flag warmstart="disable"/>
</option>
```
and checking if the behaviors persist?

pmh5050, 2023-10-14:
I've come across similar information in  [#1000](https://github.com/google-deepmind/mujoco/issues/1000), but this approach won't resolve our problem.

saran-t, 2023-10-14:
This is fairly clearly not a warmstart issue. Can you post your code?

yuvaltassa, 2023-10-14:
There is something very strange about your your code snippet. Does `i` iterate over time or over threads? I would have expected

```C
mj_resetData(model, data[i]);
for(int t = 0; t < steps; t++)
{
    mj_step(model, data[i]);
}
```

pmh5050, 2023-10-15:
I apologize for the confusion. I made a mistake when I wrote the code snippet. As you mentioned, the variable `i` iterates over threads.

pmh5050, 2023-10-15:
I've identified the cause of this issue. It is related to the internal static variable of the IK library and is not related to MuJoCo. As a result, I am closing this issue, and I sincerely apologize for any inconvenience or time that may have been taken up. :cry:

closed 2023-10-15 by pmh5050