google-deepmind/mujoco #1094: Support for Unreal Engine Plugin

Ckerrr, 2023-10-12:
**Is your feature request related to a problem? Please describe.**
N/A

**Describe the solution you'd like**
Do we have a plan to hook up with Unreal Engine as it supports Unity as a plugin today?

**Describe alternatives you've considered**
N/A

**Additional context**
Unreal Engine may have better visual fidelity compared Unity.

yuvaltassa, 2023-10-13:
Unreal is nice, but it is 60GB, last I checked. Adding it as a dependency would be a bit of a nightmare.

That said, we are actively looking into creating a bridge to external renderers, including Unreal, based on the USD specification.

closed 2023-10-13 by yuvaltassa

utkuoguzman, 2023-12-15:
Unreal Engine 5.3.2 is incredible looking and so fast. Just download the Linux binaries and try for yourself. It accepts plugins that can be turned on & off.
The Unreal Engine would be incredible for robotic application because its realistic renders are actually useful in computer vision, unlike other renderers.

utkuoguzman, 2024-04-30:
> Unreal is nice, but it is 60GB, last I checked. Adding it as a dependency would be a bit of a nightmare.
> 
> That said, we are actively looking into creating a bridge to external renderers, including Unreal, based on the USD specification.

Hi, What is your current projection on this idea? Will we see it within a couple of years? Thanks.

erez-tom, 2024-04-30:
The short-term plan is to write a script for importing assets specified as USD. The next step would be to write a Hydra delegate that exports the transforms of the different geoms for external rendering. We're not currently planning to write this Hydra bit in-house, but we'd be thrilled to offer support if someone is willing to take that on.

carTloyal123, 2025-02-19:
Would it be easier to just write an Unreal Engine Plugin? I have been working on this for a little while and am happy to share results. Basically just taking the built binaries from Mujoco and using them for physics steps instead of Unreal's built in physics. Of course the are niceties like custom meshes, clean API interface for other Unreal components etc. but first pass is just a plugin you can use for physics steps then sync the Mujoco objects with Unreal objects.

> Unreal is nice, but it is 60GB, last I checked. Adding it as a dependency would be a bit of a nightmare.
> 
> That said, we are actively looking into creating a bridge to external renderers, including Unreal, based on the USD specification.

The relationship is not a 60GB dependency...it would be an Unreal Engine plugin that uses the Mujoco binaries. Not adding more dependencies to Mujoco itself.

What Hydra?