google-deepmind/mujoco #1095: Measuring the torque in a joint with sensor?

QiyangYan, 2023-10-12:
Hi,

I'm a student and I'm trying to use MuJoCo for sim2real of a gripper.

I'm looking for some help with **measuring the torque in a joint**.

Here is my question:
I would like to measure the joint torque for the gripper, like what you did in OP3 where you measured the joint torque as a penalty. Did you add a force-torque sensor to the joint or there is a different way of measuring it? Would you mind providing me with a short example or a guide for measuring the torque of a joint? THANKS!

yuvaltassa, 2023-10-13:
You need to define more accurately what you mean by "torque in a joint". You could mean:
1. The torque generated by the actuator
2. The total torque along the joint axis due to the actuator and everything else.
3. The total 3D torque through the joint, including in axes orthogonal to the joint axis.

Which is it?

QiyangYan, 2023-10-13:
Hi, by torque in joint, I’m referring to second point. I would like to measure the torque within the joint due to external force.

yuvaltassa, 2023-10-13:
What do you mean "external force"? Not including the actuator or including the actuator?

QiyangYan, 2023-10-13:
The joint is a position-controlled joint, like the knees' joint of OP3, and I want to monitor the variation of torque generated by the actuator when an external force is applied to the joint in the opposite direction, resisting the movement. So it should include the actuator?

For example, in the paper of the [OP3](https://arxiv.org/abs/2304.13653), inside the reward, there is a penalty, equal to the magnitude of the torque measured at the player’s knees. This discourages the player from learning gaits which cause high forces on the knees, for example during ground impacts, which can damage a physical robot.

And, I'm using python.

yuvaltassa, 2023-10-13:
`data.joint("my_joint").qfrc_constraint + data.joint("my_joint").qfrc_smooth`

closed 2023-10-13 by yuvaltassa

ziqiao30, 2024-06-06:
Hi i just wornde what the difference bewteen: data.joint("my_joint").qfrc_constraint + data.joint("my_joint").qfrc_smooth;
and just put a site to the joint and use torque sensor to read the torque value. I try to plot on axis from the torque sensor and the value calculate from the data.joint, there are different scale, and some time the value are quite different(not only in different scale)

NuCapybara, 2024-07-30:
> qfrc_smooth

Hi Ziqiao, I found a post somewhat related to your question. Have you figured out?
https://www.roboti.us/forum/index.php?threads/best-way-to-represent-robots-torque-sensors.4181/

NuCapybara, 2024-07-30:
> Hi i just curiosity whats the difference bewteen: data.joint("my_joint").qfrc_constraint + data.joint("my_joint").qfrc_smooth; and just put a site to the joint and use torque sensor to read the torque value. I try to plot on axis from the torque sensor and the value calculate from the data.joint, there are different scale, and some time the value are quite different(not only in different scale)

Hi Ziqiao, I found a post somewhat related to your question. Have you figured out?
https://www.roboti.us/forum/index.php?threads/best-way-to-represent-robots-torque-sensors.4181/