google-deepmind/mujoco #1097: Wench which wrap or unwrap the cable

rishabkr29, 2023-10-13:
Hi,

I'm a student and I'm trying to use MuJoCo for my project.

I'm looking for some help with the xml file. I wanted to develop a wench which will wrap and unwrap the cable and ultimately the cable will be connected to manipulator. Basically I am working to develop the cable driven manipulator and for which I need to develop the wench which will wrap and unwrap the cable. 
[image]
 

Here is a model which explains my question:

<details>
<mujoco>
    <compiler autolimits="true" />
    <worldbody>
        <geom type="plane" size="0.8 0.8 0.1" rgba="1 1 1 1"/>
        <body name="b_winch" pos="0 0 0.1">
            <geom type="cylinder" fromto="0 0 -0.05 0 0 0.05" size="0.05" />
            <joint name="winch_0" range="-1.5708 1.5708" />
            <geom type="cylinder" fromto="0 0 -0.04 0 0 -0.05" size="0.12" />
            <geom type="cylinder" fromto="0 0 0.04 0 0 0.05" size="0.12" />
            <body name="B0" pos="0 0 0">
                <composite type="rope" count="10 1 1" spacing="0.04" offset="0 0 2">
                    <joint kind="main" damping="0.005" />
                    <geom type="capsule" size=".01 .015" rgba=".8 .2 .1 1" />
                </composite>
            </body>
            <body pos="0.5 0 0">
                <geom type="box" size=".01 .01 0.1" rgba=".2 .2 .1 1" />
                <joint name="winch_1" range="-1.5708 1.5708" />
                <site name="l1" pos="0.5 0.0025 0.1" rgba="0 .9 0 1"/> 
            </body>
        </body>
    </worldbody>

    <actuator>
        <motor name="winch_0" gear="2" joint="winch_0" />
    </actuator>
</mujoco>

Here is a screenshot illustrating my question:

rishabkr29, 2023-10-18:
<mujoco>
    <compiler autolimits="true" />
    <worldbody>
        <geom type="plane" size="0.8 0.8 0.1" rgba="1 1 1 1"/>
        <body name="b_winch" pos="0 0 0.1">
            <geom type="cylinder" fromto="0 0 -0.05 0 0 0.05" size="0.05" />
            <joint name="winch_0" range="-1.5708 1.5708" />
            <geom type="cylinder" fromto="0 0 -0.04 0 0 -0.05" size="0.12" />
            <geom type="cylinder" fromto="0 0 0.04 0 0 0.05" size="0.12" />
            <body name="B0" pos="0 0 0">
                <composite type="rope" count="10 1 1" spacing="0.04" offset="0 0 2">
                    <joint kind="main" damping="0.005" />
                    <geom type="capsule" size=".01 .015" rgba=".8 .2 .1 1" />
                </composite>
            </body>
            <body pos="0.5 0 0">
                <geom type="box" size=".01 .01 0.1" rgba=".2 .2 .1 1" />
                <joint name="winch_1" range="-1.5708 1.5708" />
                <site name="l1" pos="0.5 0.0025 0.1" rgba="0 .9 0 1"/> 
            </body>
        </body>
    </worldbody>

    <actuator>
        <motor name="winch_0" gear="2" joint="winch_0" />
    </actuator>
</mujoco>

rishabkr29, 2023-10-18:
[image]

rishabkr29, 2023-10-19:
> Hi, I think there was some issue when you pasted the model? I only see 3 lines.

Yes, I rectified it. Now please check. I wanted to wrap and unwrap the cable around winch so that the attached link can come closer.

quagla, 2023-10-25:
Hi, thanks for the updated model! Ropes are no longer supported in MuJoCo 3, I would recommend to update the model to either use cable or flex, depending on whether you would like an inextensible or extensible cable. Let me know how you get on converting your model (it should be a small change).

rishabkr29, 2023-10-26:
ultimately, I am trying to integrate #1096 with the given model I shared with you.  Here is the layout I am trying to do.
[image]

rishabkr29, 2023-10-26:
> Hi, thanks for the updated model! Ropes are no longer supported in MuJoCo 3, I would recommend to update the model to either use cable or flex, depending on whether you would like an inextensible or extensible cable. Let me know how you get on converting your model (it should be a small change).
1. I am not able to understand how I can the cable type creates a 1D chain of bodies connected with ball joints ?
2. also I want an inextensible cable, by saying cable I mean a steel wire with diameter around 3 mm.

quagla, 2023-10-30:
In principle, you only have to replace `type="rope"` with `type="cable"` and `spacing="0.04"` with something like `curve="s"`. I would remove `offset="0 0 2"` as well.

closed 2023-10-31 by quagla

rishabkr29, 2023-11-06:
> In principle, you only have to replace `type="rope"` with `type="cable"` and `spacing="0.04"` with something like `curve="s"`. I would remove `offset="0 0 2"` as well.

Thanks let me incorporate the changes.

rishabkr29, 2023-11-15:
> In principle, you only have to replace `type="rope"` with `type="cable"` and `spacing="0.04"` with something like `curve="s"`. I would remove `offset="0 0 2"` as well.

Thanks, but I am encountering issue while naming the cable and connecting it with the other body.
Here is my code for cable composites-
 
<body name="CB0" pos="-0.4 0.15 0.06" euler="0 0 0">
        <composite  type="cable" count="5 1 1" curve="s" prefix = "C" size = "0.2" initial = "ball">
          <joint kind="main" damping="0.005" />
          <geom type="capsule" size=".003 .0015" rgba=".8 .2 .1 1" />
        </composite>
    </body>

I am using equality constraint to connect the both end of the cable using - 
<equality>
    <connect name = "cable1" body1="block" body2="CB10" anchor="-0.4 0.15 0.06"  solref="-10000 -1000" />
</equality>