google-deepmind/mujoco #1100: Orientation issue for small meshes

snasiriany, 2023-10-17:
For models that have small meshes, I was seeing the issue `Error: mesh volume is too small...`

I'm able to make this issue go away by adding the following line to the body containing the meshes: `<inertial pos="0 0 0" mass="0" diaginertia="0 0 0"/>`

Adding this line resolves the error, but the offending small mesh is oriented incorrectly:
[image]

I've noted that the orientation issue goes away when I increase the `scale` parameter of the meshes to a large enough value:
[image]

Attaching a zip file with minimal models here:
[small_model_2.zip](https://github.com/google-deepmind/mujoco/files/12969309/small_model_2.zip)

To reproduce: `python -m mujoco.viewer --mjcf small_model_2/model_small_scale.xml`

I believe this is a bug.

- System information:
  - Operating system: both Mac (M1), Ubuntu
  - MuJoCo version: 2.3.7

kevinzakka, 2023-10-18:
Add `shellinertia=true` to the hood light geoms and your problem should go away.

quagla, 2023-10-24:
Why do you need to make it false? My intention was that shellinertia should only have been dependent on the mesh topology, not its size.

closed 2024-08-15 by kevinzakka