google-deepmind/mujoco #1109: The support for AMD CPUs and integrated Radeon graphics is greatly needed

Nghiattk27, 2023-10-19:
My laptop is equipped with an AMD Ryzen 7 7730U CPU and integrated AMD Radeon graphics. I'm aware of the issue regarding the inability to display the Mujoco interface (https://github.com/google-deepmind/mujoco/issues/639, https://github.com/google-deepmind/mujoco/issues/894), and I've resorted to using an alternative solution (https://github.com/rohanpsingh/mujoco-python-viewer). While this workaround is temporary, I still hope for support from the Mujoco development team. However, when I updated to Mujoco 3.0.0, it seems that this temporary solution is no longer effective, and the problems with displaying the user interface persist for AMD's integrated chip and graphics card. I don't believe this is an insurmountable issue; it might be a coding error causing the interface display problem. Nevertheless, I am hopeful for a solution that would allow me to use Mujoco on my laptop.

yuvaltassa, 2023-10-19:
We are in the process of acquiring an AMD-equipped machine in order to reproduce and hopefully finally resolve this issue

Nghiattk27, 2023-10-20:
I'm very glad to hear that. I hope the issue will be resolved soon for the AMD user community

saran-t, 2023-10-20:
Let's track progress in https://github.com/google-deepmind/mujoco/issues/639, I've asked for a bit of help debugging over there. I'm reasonably hopeful that we'll finally get this fixed!

closed 2023-10-20 by saran-t