google-deepmind/mujoco #1114: Camera frustum: Remove shadow, specify frustum depth

vikashplus, 2023-10-21:
Thanks for the camera frustum visualization feature. It's very informative, especially with moving cameras to ensure the subject is in the field of view. I'm wondering if it's possible to 
1. Change the depth of rendered frustum. It currently seems to extend to infinity. The frustum starts overlapping with scenes with multiple cameras. If the depth can be specified (maybe in visual/map) that will greatly improve the usability. The ability to change frustum color can also be helpful (visual/rgba)
2. Is it possible to remove the shadow of the camera's frustum rendering? Below is an example of the same scene with and without rendering. The shadows on the table make the scene very dark to operate. (I understand that playing around with the position of the light can help here. It might not be a watertight solution as lights are usually above cameras in most scenes and this issue will likely be quite common)
[image]
[image]

Here is the model for the above scene 
```xml
<mujoco>
  <worldbody>
    <light pos="0 0 5"/>
    <geom type="plane" size="10 10 .01"/>
    <geom type="box" size=".375 .6 .01" pos="0 0 .5" rgba=".2 .5 .6 1"/>
    <geom type="sphere" size=".2" pos="0 0 2" rgba=".3 .6 .2 1"/>

    <camera pos="1.5 0 1.5" xyaxes="0 -1 0 1 0 -1" focalpixel="1920 1200" resolution="1920 1200" sensorsize="9.6e-3 6e-3"/>
    <camera pos="-1.5 0 2.5" xyaxes="0 -1 0 .3 0 1" focalpixel="1920 1200" resolution="1920 1200" sensorsize="9.6e-3 6e-3"/>
  </worldbody>
</mujoco>
```

saran-t, 2023-10-21:
@quagla it should be straightforward to make the frustum not cast shadows

quagla, 2023-10-23:
Hi @vikashplus the depth of the frustum can be set with visual/zfar (which of course affects the rendering too).

I will soon submit the fix for the shadows.

vikashplus, 2023-10-23:
zfar is a global. Setting it low will clip the scene rendering as well. it might not be beneficial to couple frustum annotation size with scene rendering.
 
 Do you have suggestions on how one can go about getting visuals like these?
[image]
[image]

vikashplus, 2023-10-23:
@quagla - while you are on shadows, it will be great to disable shadows cast from all annotations (cameras, lights, etc). Especially lights as they cast shadow exactly where the light is supposed to hit defeating the whole purpose of adding a light :)

yuvaltassa, 2023-10-23:
@vikashplus, what do you think would be the most convenient way to specify frustum depth? How about the following, where `x` is scaled by `meansize` (like all the other numbers in `<scale>`)?

```xml
<visual>
  <scale frustum="x"/>
</visual>
```

vikashplus, 2023-10-23:
I like the idea. Building on top of what you mentioned, maybe something like this?
``` xml 
<visual>
  <scale frustum_depth="depth U V"/> <!--U, V are optional, and will used if no other frustum details (focalpixel, resolution, sensorsize) are provided -->
  <rgba frustum_plane="r g b a" frustum_edge="r g b a"/>
</visual>

```

closed 2023-10-25 by copybara-service[bot]