google-deepmind/mujoco #1116: Python Viewer: Clear displayed error post successful model load

vikashplus, 2023-10-21:
1. Load a healthy model on the viewer using `python -m mujoco.viewer --mjcf=model.xml`
2. Now edit the model.xml to introduce a bug
3. Reload the viewer. The viewer correctly shows the compilation error 
4. Now undo/fix the model edit/error
5. Hit reload on the viewer to pick the new model. 
    - As there is no error, the model loads just fine
    - The old error from step 3 is on display. If my understanding is correct - there shouldn't be any error message on the display as the model load was successful 
[image]

Version: MuJoCo 3.9
OS: Ubuntu 18.04

saran-t, 2023-10-21:
Can you check if this issue also happens with the native simulate app?

kevinzakka, 2023-10-23:
@saran-t Native simulate reloads the model fine and the error message disappears.

vikashplus, 2023-10-23:
@kevinzakka beats me to this. 
Thanks Kevin :)

commit af388452 (2024-03-22): [Python Viewer] clear load error on successful reload, fixes #1116
PR #1534 (2024-03-22): [Python Viewer] clear load error on successful reload, fixes #1116
This has been annoying me lately, turns out it is a one line fix. Fixes #1116.

[`main.cc`](https://github.com/google-deepmind/mujoco/blob/67982c4961b47a899ee93e988989c0bf2687b72f/simulate/main.cc#L236) always overwrites the `simulate.load_error` on a successful load, which implicitly sets an empty string if the load is successful. The python viewer's `viewer.py` just needs to do the same.
closed 2024-03-22 by copybara-service[bot]