google-deepmind/mujoco #1121: Incorrect Matching between Depth and Segmentation Renders

Edu4000, 2023-10-23:
Hi,

I'm a student and I'm trying to use MuJoCo's Python bindings for multiple object recognition using point clouds generated from depth renders.

I'm looking for some help with figuring out how to precisely filter a depth render with a corresponding segmentation render.

Currently I am generating a depth and a segmentation render from a single camera, and using the segmentation render to separate the depth readings depending on the number of objects detected, en short it is a 1 to 1 match of segmentation with depth value.

The issue with this approach is that in the edges of a figure, the match between the depth and segmentation render might fail. (e.g. I get the depth of object A, but assign the id of object B)

I am running everything on a JupyterNotebook:
- Python    - v. 3.10.13
- Mujoco    - v. 3.0.0
- Open3D  - v. 0.17.0
- NumPy    - v. 1.26.0
- Mediapy  - v. 1.1.9

Here is an example which explains my question:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco>
  <visual>
    <global offwidth="680" offheight="420"/>
  </visual>

  <worldbody>
    <geom name="floor" type="plane" size="2 2 0.1" rgba="0.9 0.9 0.9 1"/>
    <geom name="box" type="box" pos="0 0 0.2" size="0.2 0.2 0.2" rgba="0.9 0.1 0.1 1"/>
    <camera name="camera2" pos="-0.8 -0.8 1" quat="-0.82 -0.42 0.17 0.35" />
  </worldbody>
</mujoco>
```

</details>

<details>
<summary>Method to get the point cloud</summary>

```python
def get_point_cloud(frame, sgmnt, w, h):
    fovy = 60
    f_y = 0.5 * h / np.tan(fovy * np.pi / 360)
    f_x = f_y * (w / h)

    # Create arrays for x and y coordinates
    x = (np.arange(0, w) - w/2) * w / h / f_x
    y = (np.arange(0, h) - h/2) / f_y
    z = np.ones((w*h, 1))
    
    # Use meshgrid to create two grids of x and y coordinates
    X, Y = np.meshgrid(x, y)

    # Stack X and Y to create the matrix
    matrix = np.vstack((X.flatten(), Y.flatten())).T

    # Vectorized matrix to convert depth to 3D points
    matrix = np.concatenate((matrix, z), axis=1)

    frame = frame.reshape((-1, 1))
    sgment = sgmnt.reshape((-1, 1))

     # Get the 3D points from depth rendering
    points = frame * matrix

    # Discarding NaN elements
    sgment = sgment[~np.isnan(points).any(axis=1)]
    points = points[~np.isnan(points).any(axis=1)]

    return points, sgment
```

</details>

```python
import mujoco, mujoco.renderer, mujoco.viewer
import numpy as np
import open3d as o3d

# Creating model and data
m = mujoco.MjModel.from_xml_string(xml)
d = mujoco.MjData(m)

# Creating the renderers
depth_renderer = mujoco.Renderer(m, 420, 680); depth_renderer.enable_depth_rendering()
sgmnt_renderer = mujoco.Renderer(m, 420, 680); sgmnt_renderer.enable_segmentation_rendering()

# Step and render
mujoco.mj_step(m, d)

depth_renderer.update_scene(d, 0)
sgmnt_renderer.update_scene(d, 0)

# Rendering depth and segmentation frames
depth = depth_renderer.render()
sgmnt = sgmnt_renderer.render()[:, :, 0].astype("float")

depth[depth > 4] = np.nan
sgmnt[sgmnt == -1] = np.nan

# Generating point clouds
point_cloud, ids = get_point_cloud(depth, sgmnt, 680, 420)

# Displaying point clouds
pcd = o3d.geometry.PointCloud()
for id in np.unique(ids):
    pcd.points = o3d.utility.Vector3dVector(point_cloud[(ids == id).any(axis=1)])
    o3d.visualization.draw_geometries([pcd])

```

Here are some screenshots showing what is being rendered and the result I am getting.

<details>
<summary>This is the scene that is being rendered with depth and segmentation correspondingly.</summary>

[image]

</details>

<details>
<summary>And this is the reconstruction of the two objects present. (The floor and the cube)</summary>

[image]

[image]

[image]

</details>

yuvaltassa, 2023-10-23:
I suspect that your depth renders are using antialiasing, which creates ambiguity. Try setting [offsamples](https://mujoco.readthedocs.io/en/latest/XMLreference.html#visual-quality-offsamples) to 0.

Edu4000, 2023-10-23:
Thank you, I think that was it. With that setting now it is working correctly.

```XML
<mujoco>
  <visual>
    <quality offsamples="0"/>
    <global offwidth="680" offheight="420"/>
  </visual>

  <worldbody>
    <geom name="floor" type="plane" size="2 2 0.1" rgba="0.9 0.9 0.9 1"/>
    <geom name="box" type="box" pos="0 0 0.2" size="0.2 0.2 0.2" rgba="0.9 0.1 0.1 1"/>
    <camera name="camera2" pos="-0.8 -0.8 1" quat="-0.82 -0.42 0.17 0.35" />
  </worldbody>
</mujoco>
```

closed 2023-10-23 by Edu4000