google-deepmind/mujoco #1124: Solimp and solref related help

Aditya-1729, 2023-10-23:
Hi,

I'm a student and I'm trying to use MuJoCo for a wiping task with Panda robot

I'm looking for some help with setting up the contact dynamics. I dont intend to dig deep into the theory behind how the solver uses these parameters. But having spent a little bit of time, I am afraid, I wasnt able comprehend the mujoco documentation much. I am struggling to find resources that point to some good ballpark figures for solimp and solref, which would  correspond to some catagerocial behaviour. For example, I want to model a contact between a solid slab and a 3d printed end effector (PLA) with a flat face. Can somebody please comment on whether these figures are realistic, or may be point me to some relevant examples/material? On the slab xml, I dont specify the solimp and solref values; for the gripper, this is how the file looks like: 

Here is a model which explains my question:
<mujoco model="wiping_hand">
    <asset>
        
        <mesh name="wipe_eef" file="meshes/panda_wipe/wipe_eef.stl" scale="0.001 0.001 0.001" />
        
        
    </asset>
    <worldbody>
        <body name="wiping_gripper" pos="0 0 0.0494" quat="0.707107 0 0 -0.707107" >

            <site name="ft_frame" pos="0 0 0" size="0.0001" rgba="1 1 0 0" type="sphere" group="1"/>
            <inertial pos="0.00 0.001 -0.018" mass="0.709" diaginertia="55e-8 54e-8 53e-8"/>

            <geom type="mesh" pos="0 0 0.0" name="wiping_surface_vis" mesh="wipe_eef" contype="1" group="1" solimp="0.2 0.9 0.01" solmix="10000" solref="0.05 1" rgba="0.25 0.25 0.25 1" />
            <geom type="sphere" group="0" pos="0 0.025 0.0536" size="0.001" name="wiping_corner1" solimp="0.0 0.1 0.01" solmix="10000" solref="0.1 3" rgba="1 0 0 1" friction="0.001 0.005 0.0001"/>            
            <site type="sphere" rgba="1 0 0 1" pos="0 0.025 0.0536" size="0.001" name="wiping_corner1_site"/>

            <geom type="sphere" group="0" pos="0.025 0.0 0.0536" size="0.001" name="wiping_corner2" solimp="0.0 0.5 0.01" solmix="10000" solref="0.1 3" rgba="0 1 0 1" friction="0.001 0.005 0.0001"/>
            <site type="sphere" rgba="0 1 0 1" pos="0.025 0.0 0.0536" size="0.001" name="wiping_corner2_site"/>
                                    
            <geom type="sphere" group="0" pos="-0.025 0.0 0.0536 " size="0.001" name="wiping_corner3" solimp="0.0 0.5 0.01" solmix="10000" solref="0.1 3" rgba="0 0 1 1" friction="0.001 0.005 0.0001"/>
            <site type="sphere" rgba="0 0 1 1" pos="-0.025 0.0 0.0536 " size="0.001" name="wiping_corner3_site"/>
 
            <geom type="sphere" group="0" pos="0.0 -0.025 0.0536" size="0.001" name="wiping_corner4" solimp="0.0 0.5 0.01" solmix="10000" solref="0.1 3" rgba="1 1 0 1" friction="0.001 0.005 0.0001"/>
            <site type="sphere" rgba="1 1 0 1" pos="0.0 -0.025 0.0536" size="0.001" name="wiping_corner4_site"/>
                                    
 
            
            <body name="eef" pos="0 0 0.0536" quat="1 0 0 0"> 
                <site name="grip_site" pos="0 0 0" size="0.001 0.001 0.001" rgba="1 0 0 0.5" type="sphere" group="1"/>
                <site name="ee_x" pos="0 0 0" size="0.005 .1"  quat="0.707105  0 0.707108 0 " rgba="1 0 0 0" type="cylinder" group="1"/>
                <site name="ee_y" pos="0 0 0" size="0.005 .1" quat="0.707105 0.707108 0 0" rgba="0 1 0 0" type="cylinder" group="1"/>
                <site name="ee_z" pos="0 0 0" size="0.005 .1" quat="1 0 0 0" rgba="0 0 1 0" type="cylinder" group="1"/>
                
                <site name="grip_site_cylinder" pos="0 0 0" size="0.005 10" rgba="0 1 0 0.3" type="cylinder" group="1"/>
            </body>
        </body>
    </worldbody>
    <sensor>
        <force name="force_ee" site="ft_frame"/>
        <torque name="torque_ee" site="ft_frame"/>
        <touch name="touch1" site="wiping_corner1_site"/>
        <touch name="touch2" site="wiping_corner2_site"/>
        <touch name="touch3" site="wiping_corner3_site"/>
        <touch name="touch4" site="wiping_corner4_site"/>
    </sensor>
</mujoco>

saran-t, 2023-10-24:
@yuvaltassa might be able to eyeball it but really if you want to reproduce the behaviour of specific materials your best bet is to make force measurements on the real system and do system identification.

yuvaltassa, 2023-10-25:
As @saran-t says, the only way to do this is to use a force sensor and do some system identification.

A couple of notes regarding your XML:
1. `solref[0] = 0.1` (100ms) is really soft.
2. You can use [priority](https://mujoco.readthedocs.io/en/latest/XMLreference.html#body-geom-priority) instead of a really high solmix.

closed 2023-10-25 by yuvaltassa

Aditya-1729, 2023-10-28:
@yuvaltassa Thank you for your response. 
I am trying to specifiy the soref in negative form as it is intended to be more intuitive. Can you tell me the documentation implies the following logic:
1mm of penetration with 5N of normal force implies:
 k = 5000
 And I need to specify the -stiffness in solref, so I can choose d_width=0.1, which leads to:
 stiffness= k x (d_width^2) = 50.

yuvaltassa, 2023-10-28:
No, that is wrong in multiple ways.

Perhaps the docs need to be improved further. Before I explain more, I need to better understand what you want. Say a ball is on the floor and it weighs 5N. Do you want 
1. The penetration **at rest** should be 1mm.
2. The maximum penetration **upon impact** should be no more than 1mm.

Aditya-1729, 2023-10-29:
Alright. Thank you for pointing that out. 
So I want the penetration to be 1mm **at rest** with 5N of force.