google-deepmind/mujoco #1127: Clarify "pos" documentation for body/joint.

joemasterjohn, 2023-10-24:
We came across some documentation confusion when fixing a bug in our [parser for the MuJoCo XML format for Drake](https://github.com/RobotLocomotion/drake/blob/master/multibody/parsing/detail_mujoco_parser.cc) during [a recent review](https://reviewable.io/reviews/RobotLocomotion/drake/20414). Specifically the `pos` attribute of the `body/joint` tag.

The MuJoCo doc reads:
> pos: real(3), “0 0 0”
> Position of the joint, specified in the frame of the parent body. For free joints this attribute is ignored.

We interpreted the frame specified in this attribute to be the body frame of the parent body of the joint. But after comparing kinematics between what Drake parsed and MuJoCo we found that the frame specified is the body frame of the "child body" of the joint. i.e. interpreting "parent body" in the doc string to mean the body corresponding to the <body> xml node that is the parent node of this <joint> xml node.

We found that language to be a bit confusing and led to our misunderstanding, so I am suggesting to clarify the docs. It seems in doc strings for other attributes of `body/joint` "body frame" is used for the frame in question (correct me if I'm wrong). My suggestion would be to use that same term for `pos`:
> pos: real(3), “0 0 0”
> Position of the joint, specified in the body frame. For free joints this attribute is ignored.

Thanks!

saran-t, 2023-10-24:
@yuvaltassa is handling this, apparently there are more places with this ambiguity and some occurrences are trickier to fix than this one.

closed 2023-10-25 by copybara-service[bot]

yuvaltassa, 2023-10-27:
@joemasterjohn this is now fixed on the Latest docs branch, please let us know if anything like this comes up in future.
https://mujoco.readthedocs.io/en/latest/XMLreference.html

commit 190bfac2 (2023-10-25): Rephrase documentation in places where "parent body" creates ambiguity between "the body which contains the element" and "the parent body of the body which contains the element".

Fixes #1127

PiperOrigin-RevId: 576486971
Change-Id: I54d0c351e95101c5f7907a3c5bb6be18904c03ad