google-deepmind/mujoco #1128: Fail to "pip install mujoco" on M1 chip MacOS

SiboD, 2023-10-24:
I tried to install mujoco using "pip install mujoco" and met the following issue:

Chip: Apple M1
macOS: Ventura 13.5.2 
Python version: same issue from 3.8 to 3.11
Virtual env: Anaconda

Building wheels for collected packages: mujoco
  Building wheel for mujoco (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: /opt/anaconda3/bin/python /opt/anaconda3/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py build_wheel /var/folders/s1/wq0qnd055hnf9j4w9wkpgc1r0000gn/T/tmp6s2r082q
       cwd: /private/var/folders/s1/wq0qnd055hnf9j4w9wkpgc1r0000gn/T/pip-install-sokt20v0/mujoco_988befba4cdc49cb8d25398189d34a99
  Complete output (77 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.macosx-10.9-x86_64-cpython-38
  creating build/lib.macosx-10.9-x86_64-cpython-38/mujoco
  copying mujoco/viewer.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
  copying mujoco/renderer.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
  copying mujoco/renderer_test.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
  copying mujoco/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
  copying mujoco/msh2obj_test.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
  copying mujoco/msh2obj.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
  copying mujoco/rollout.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
  copying mujoco/rollout_test.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
  copying mujoco/bindings_test.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
  copying mujoco/viewer_test.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
  copying mujoco/render_test.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
  copying mujoco/gl_context.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco
  creating build/lib.macosx-10.9-x86_64-cpython-38/mujoco/cgl
  copying mujoco/cgl/cgl.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco/cgl
  copying mujoco/cgl/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco/cgl
  creating build/lib.macosx-10.9-x86_64-cpython-38/mujoco/egl
  copying mujoco/egl/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco/egl
  copying mujoco/egl/egl_ext.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco/egl
  creating build/lib.macosx-10.9-x86_64-cpython-38/mujoco/osmesa
  copying mujoco/osmesa/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco/osmesa
  creating build/lib.macosx-10.9-x86_64-cpython-38/mujoco/glfw
  copying mujoco/glfw/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco/glfw
  creating build/lib.macosx-10.9-x86_64-cpython-38/mujoco/testdata
  copying mujoco/testdata/msh.xml -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco/testdata
  copying mujoco/testdata/abdomen_1_body.msh -> build/lib.macosx-10.9-x86_64-cpython-38/mujoco/testdata
  running build_ext
  Traceback (most recent call last):
    File "/opt/anaconda3/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py", line 280, in <module>
      main()
    File "/opt/anaconda3/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py", line 263, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "/opt/anaconda3/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py", line 204, in build_wheel
      return _build_backend().build_wheel(wheel_directory, config_settings,
    File "/private/var/folders/s1/wq0qnd055hnf9j4w9wkpgc1r0000gn/T/pip-build-env-2spaa3uj/overlay/lib/python3.8/site-packages/setuptools/build_meta.py", line 434, in build_wheel
      return self._build_with_temp_dir(
    File "/private/var/folders/s1/wq0qnd055hnf9j4w9wkpgc1r0000gn/T/pip-build-env-2spaa3uj/overlay/lib/python3.8/site-packages/setuptools/build_meta.py", line 419, in _build_with_temp_dir
      self.run_setup()
    File "/private/var/folders/s1/wq0qnd055hnf9j4w9wkpgc1r0000gn/T/pip-build-env-2spaa3uj/overlay/lib/python3.8/site-packages/setuptools/build_meta.py", line 341, in run_setup
      exec(code, locals())
    File "<string>", line 334, in <module>
    File "/private/var/folders/s1/wq0qnd055hnf9j4w9wkpgc1r0000gn/T/pip-build-env-2spaa3uj/overlay/lib/python3.8/site-packages/setuptools/__init__.py", line 103, in setup
      return distutils.core.setup(**attrs)
    File "/private/var/folders/s1/wq0qnd055hnf9j4w9wkpgc1r0000gn/T/pip-build-env-2spaa3uj/overlay/lib/python3.8/site-packages/setuptools/_distutils/core.py", line 185, in setup
      return run_commands(dist)
    File "/private/var/folders/s1/wq0qnd055hnf9j4w9wkpgc1r0000gn/T/pip-build-env-2spaa3uj/overlay/lib/python3.8/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
      dist.run_commands()
    File "/private/var/folders/s1/wq0qnd055hnf9j4w9wkpgc1r0000gn/T/pip-build-env-2spaa3uj/overlay/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
      self.run_command(cmd)
    File "/private/var/folders/s1/wq0qnd055hnf9j4w9wkpgc1r0000gn/T/pip-build-env-2spaa3uj/overlay/lib/python3.8/site-packages/setuptools/dist.py", line 989, in run_command
      super().run_command(command)
    File "/private/var/folders/s1/wq0qnd055hnf9j4w9wkpgc1r0000gn/T/pip-build-env-2spaa3uj/overlay/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
      cmd_obj.run()
    File "/private/var/folders/s1/wq0qnd055hnf9j4w9wkpgc1r0000gn/T/pip-build-env-2spaa3uj/normal/lib/python3.8/site-packages/wheel/bdist_wheel.py", line 364, in run
      self.run_command("build")
    File "/private/var/folders/s1/wq0qnd055hnf9j4w9wkpgc1r0000gn/T/pip-build-env-2spaa3uj/overlay/lib/python3.8/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
      self.distribution.run_command(command)
    File "/private/var/folders/s1/wq0qnd055hnf9j4w9wkpgc1r0000gn/T/pip-build-env-2spaa3uj/overlay/lib/python3.8/site-packages/setuptools/dist.py", line 989, in run_command
      super().run_command(command)
    File "/private/var/folders/s1/wq0qnd055hnf9j4w9wkpgc1r0000gn/T/pip-build-env-2spaa3uj/overlay/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
      cmd_obj.run()
    File "/private/var/folders/s1/wq0qnd055hnf9j4w9wkpgc1r0000gn/T/pip-build-env-2spaa3uj/overlay/lib/python3.8/site-packages/setuptools/_distutils/command/build.py", line 131, in run
      self.run_command(cmd_name)
    File "/private/var/folders/s1/wq0qnd055hnf9j4w9wkpgc1r0000gn/T/pip-build-env-2spaa3uj/overlay/lib/python3.8/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
      self.distribution.run_command(command)
    File "/private/var/folders/s1/wq0qnd055hnf9j4w9wkpgc1r0000gn/T/pip-build-env-2spaa3uj/overlay/lib/python3.8/site-packages/setuptools/dist.py", line 989, in run_command
      super().run_command(command)
    File "/private/var/folders/s1/wq0qnd055hnf9j4w9wkpgc1r0000gn/T/pip-build-env-2spaa3uj/overlay/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
      cmd_obj.run()
    File "<string>", line 152, in run
    File "<string>", line 166, in _find_mujoco
  RuntimeError: MUJOCO_PATH environment variable is not set
  ----------------------------------------
  ERROR: Failed building wheel for mujoco
Failed to build mujoco
ERROR: Could not build wheels for mujoco which use PEP 517 and cannot be installed directly

I tried to download the Mujoco.app and add the MUJOCO_PATH, but it then had the issue: "RuntimeError: MUJOCO_PLUGIN_PATH environment variable is not set."

kevinzakka, 2023-10-24:
Are you sure the Anaconda installation you have is for M1 and not Intel?

SiboD, 2023-10-25:
yes, it's on M1 MBP

kevinzakka, 2023-10-25:
`/opt/anaconda3/` is suspicious, that looks like an Intel install.

saran-t, 2023-10-25:
The build log clearly says `macosx-10.9-x86_64-cpython-38`. You've installed Anaconda incorrectly.

SiboD, 2023-10-25:
Thanks for pointing out! I installed the correct version of Anaconda and it's working. 

But I am curious why I have been using this wrong Anaconda for years without any other issues.

closed 2023-10-25 by kevinzakka

saran-t, 2023-10-25:
[Rosetta](https://developer.apple.com/documentation/apple-silicon/about-the-rosetta-translation-environment) magic is hiding the translation. You've inadvertently been running code slower than they should run.