google-deepmind/mujoco #1129: [MJX] Simulating models with equality constraints

heleiduan, 2023-10-24:
I'm looking for some help with simulating models with equality constraints via MJX pipeline. The model is from [mujoco_menagerie](https://github.com/google-deepmind/mujoco_menagerie/tree/main/agility_cassie). 

Here is a video, illustrating the equality constraint behaviors when setting `data.ctrl` zeros. 

https://github.com/google-deepmind/mujoco/assets/4071750/664b2215-1e50-40bb-aeef-2221cf423259

Here is a minimal code to reproduce the video. Any hints on how to resolve this? The same model would work well on the regular mujoco pipeline. Thank you!

```
import jax
from jax import numpy as jp
import numpy as np
import mediapy as media
import mujoco
from mujoco import mjx

# NOTE: Have to change geom condim from 1 to 3; comment out the `home` keyframe at the bottom of XML
xml_path = './agility_cassie/scene.xml'
model = mujoco.MjModel.from_xml_path(xml_path)
data = mujoco.MjData(model)
renderer = mujoco.Renderer(model)
mjx_model = mjx.device_put(model)

def get_image(state) -> np.ndarray:
    d = mujoco.MjData(model)
    mjx.device_get_into(d, state)
    mujoco.mj_forward(model, d)
    renderer.update_scene(d, camera='track')
    return renderer.render()

jit_step = jax.jit(mjx.step)
mjx_data = mjx.make_data(mjx_model)

images = []
for i in range(1000):
    ctrl = jp.zeros(model.nu)
    mjx_data = mjx_data.replace(ctrl=ctrl)
    mjx_data = jit_step(mjx_model, mjx_data)
    images.append(get_image(mjx_data))

media.show_video(images)
```

HyunyoungJung, 2023-10-26:
I am having the same issue with agile robotics digit. 
When I increase the solver iteration, the constraints are computed fairly well but if I bring it down to 10 iterations or so, the connection constraints are broken as in your case.

https://github.com/google-deepmind/mujoco/assets/52847647/9441e2a1-c83e-4072-b124-78d3091c0c82

https://github.com/google-deepmind/mujoco/assets/52847647/e06243b5-01fe-41b6-b272-d06f51c4e4e9

erikfrey, 2023-10-26:
Thanks for raising this issue - there was indeed a bug in the MJX solver for equality constraints.  Pushing a fix shortly.

closed 2023-10-27 by copybara-service[bot]

heleiduan, 2023-11-09:
@erikfrey We compiled the latest commit d0cabcf7dbf004f234bc0d9ba9dfe719a97d4e66 and it seems like the simulation with MJX on the same model is not behaving properly again? Could you help us to check if any other changes made to the latest that potentially makes this error? We used the same script as posted above. Thank you!

alberthli, 2023-11-12:
@erikfrey +1 on this issue.

I've also built the latest commit ([d0cabcf](https://github.com/google-deepmind/mujoco/commit/d0cabcf7dbf004f234bc0d9ba9dfe719a97d4e66)) from source and am getting errors even trying to load the [`equality.xml` test file](https://github.com/google-deepmind/mujoco/blob/d0cabcf7dbf004f234bc0d9ba9dfe719a97d4e66/mjx/mujoco/mjx/test_data/equality.xml#L4) into `mjx`. Quick MWE:
```
import mujoco
from mujoco import mjx
model = mujoco.MjModel.from_xml_path("equality.xml")
mjx_model = mjx.device_put(model)  # NotImplementedError: {2} not implemented.
```

erikfrey, 2023-11-14:
Hi @heleiduan - I'm seeing correct behavior of Agility Cassie on my end.  Can you describe the improper behavior?

We did make MJX more strict in raising errors for unsupported features, `cassie.xml` loads in MJX with the following changes:

```diff
   <default>
-    <geom contype="0" conaffinity="0" condim="1" solref="0.005 1"/>
+    <geom contype="0" conaffinity="0" condim="3" solref="0.005 1"/>
     <equality solref="0.005 1"/>
     <default class="cassie">
       <geom material="cassie" group="2"/>
@@ -60,8 +60,8 @@
         fullinertia="0.085821 0.049222 0.08626 1.276e-05 -0.00016022 -0.000414"/>
       <freejoint/>
       <geom type="mesh" mesh="cassie-pelvis" euler="0 0 90"/>
-      <!-- <geom type="sphere" size="0.15" pos="0.02 0 0.02" class="collision"/> -->
-      <geom type="ellipsoid" size="0.15 0.12 0.13" pos="0.03 0 0.03" class="collision"/>
+      <geom type="sphere" size="0.15" pos="0.02 0 0.02" class="collision"/>
+      <!-- <geom type="ellipsoid" size="0.15 0.12 0.13" pos="0.03 0 0.03" class="collision"/> -->
       <geom size="0.095" fromto="-.04 0.14 .01 .02 0.14 .01" class="collision"/>
       <geom size="0.095" fromto="-.04 -.14 .01 .02 -.14 .01" class="collision"/>
       <site name="imu" size="0.01" pos="0.03155 0 -0.07996"/>
```

heleiduan, 2023-11-15:
@erikfrey Thank you for confirming the model changes. We tried to simulate the same model (with the same changes you mentioned) and save the video using the script in this thread above. The behavior shows the closed-loop of the robot is violating equality constraint as the simulation steps. And this behaviors is the same as posted in the video above. Any clue where we should look into? Thank you!

commit 930c707e (2023-10-27): Ensure equality constraints are active irrespective of sign.

Fixes #1129

PiperOrigin-RevId: 577337939
Change-Id: Id835bb6d361525e5cb26f7696038d73d3c1a703e