google-deepmind/mujoco #1130: Strange joint limit behavior

mishmish66, 2023-10-25:
Hello! I think I've isolated a bug in mjx. I think when using joint constraints it affects other joints in strange ways.

Here are two mujoco setups

1.
`import jax
import mujoco
from mujoco import mjx

XML = r"""
    <mujoco>
        <worldbody>
            <body name="wheel" pos="0.5 0 0.75">
                <joint name="wheel_axis" axis="0 0 1" />
                <geom type="capsule" size="0.2 0.1" />
                <inertial pos="0 0 0" mass="0.5" diaginertia="0.01 0.01 0.02" />
            </body>
            <body name="block" pos="-0.5 0 0.75">
                <joint name="other_axis" axis="0 0 1" range="-0.5 0.5" />
                <geom type="capsule" size="0.2 0.1" />
                <inertial pos="0 0 0" mass="0.5" diaginertia="0.01 0.01 0.02" />
            </body>
        </worldbody>
    </mujoco>
    """

model = mujoco.MjModel.from_xml_string(XML)
mjx_model = mjx.device_put(model)

def scanf(carry, _):
    qpos, qvel = carry

    mjx_data = mjx.make_data(mjx_model)
    mjx_data = mjx_data.replace(qvel=qvel, qpos=qpos)
    next_data = mjx.step(mjx_model, mjx_data)

    return (next_data.qpos, next_data.qvel), next_data.qpos

(final_pos, final_vel), poss = jax.lax.scan(
    scanf,
    (jax.numpy.array([0.0, 0.0]), jax.numpy.array([100.0, 0.0])),
    None,
    1024,
)

print(poss)`

2.

`import jax
import mujoco
from mujoco import mjx

XML = r"""
    <mujoco>
        <worldbody>
            <body name="wheel" pos="0.5 0 0.75">
                <joint name="wheel_axis" axis="0 0 1" />
                <geom type="capsule" size="0.2 0.1" />
                <inertial pos="0 0 0" mass="0.5" diaginertia="0.01 0.01 0.02" />
            </body>
            <body name="block" pos="-0.5 0 0.75">
                <joint name="other_axis" axis="0 0 1" />
                <geom type="capsule" size="0.2 0.1" />
                <inertial pos="0 0 0" mass="0.5" diaginertia="0.01 0.01 0.02" />
            </body>
        </worldbody>
    </mujoco>
    """

model = mujoco.MjModel.from_xml_string(XML)
mjx_model = mjx.device_put(model)

def scanf(carry, _):
    qpos, qvel = carry

    mjx_data = mjx.make_data(mjx_model)
    mjx_data = mjx_data.replace(qvel=qvel, qpos=qpos)
    next_data = mjx.step(mjx_model, mjx_data)

    return (next_data.qpos, next_data.qvel), next_data.qpos

(final_pos, final_vel), poss = jax.lax.scan(
    scanf,
    (jax.numpy.array([0.0, 0.0]), jax.numpy.array([100.0, 0.0])),
    None,
    1024,
)

print(poss)`

These are exactly the same, the only difference is that in the first one the second joint is limited.
When I run these though, the first joint should just spin like crazy unresisted forever in both cases right?

Instead in 1 we get:

[[ 2.0000000e-01  0.0000000e+00]
 [ 3.5800007e-01  0.0000000e+00]
 [ 4.8082006e-01  0.0000000e+00]
 ...
 [-3.0173851e-17  0.0000000e+00]
 [ 3.6536127e-15  0.0000000e+00]
 [ 2.9679528e-17  0.0000000e+00]]
 
 and in 2 we get:
 
[[2.0000000e-01 0.0000000e+00]
 [4.0000001e-01 0.0000000e+00]
 [6.0000002e-01 0.0000000e+00]
 ...
 [2.0439803e+02 0.0000000e+00]
 [2.0459802e+02 0.0000000e+00]
 [2.0479802e+02 0.0000000e+00]]

Sorry if this is not a bug, or if this is already known!

btaba, 2023-10-25:
Hi @mishmish66 , we pushed out a couple fixes relating to this bug. 843360b and dd841c9

Can you see if the state at HEAD works for you?

closed 2023-11-10 by btaba

mishmish66, 2023-12-24:
> Hi @mishmish66 , we pushed out a couple of fixes relating to this bug. [843360b](https://github.com/google-deepmind/mujoco/commit/843360b73241c0b506c42ccf6c27027f81ae8cfa) and [dd841c9](https://github.com/google-deepmind/mujoco/commit/dd841c92831fd470c5570561efe07036a260efd1)
> 
> Can you see if the state at HEAD works for you?

Hey! I got totally swamped with schoolwork, but now that the semester is over I can confirm that with the current version 3.1.1 on pypy this is fixed. Thank you!