google-deepmind/mujoco #1134: Poor Performance with MJX on Modern GPU

SumeetBatra, 2023-10-26:
Hi folks,

I'm trying out the MJX pipeline for RL training following the example in the collab notebook uploaded with the 3.0 release. The env and environment definitions are pretty much the same (humanoid env) as the example, but I'm mixing in some brax wrappers (jax2torch, etc) for convenience. I'm running 2048 parallel envs with an RTX 3090 and only getting ~60k fps. In contrast, I've trained models with brax simulation at 200-300k fps with the same hardware and I'm not sure what's causing the discrepancy. My understanding is the forward dynamics propagation and # of contacts for pairwise collision checking is roughly the same in MJX and brax, so that shouldn't result in any significant fps discrepancy, although do correct me if I'm wrong. Perhaps it is something on my end? I did have to create my own gymwrapper for the humanoid env, since the brax VectorizedGymWrapper doesn't natively work with MjxEnv:

```
class GymWrapper(gym.Env):
    def __init__(self, env: Env, seed: int = 0, backend: Optional[str] = None):
        self._env = env
        self.metadata = {
            'render.modes': ['human', 'rgb_array'],
            'video.frames_per_second': 1 / self._env.dt
        }
        self.backend = backend
        self._key = None
        self.seed(seed)
        self._state = None

        def reset(key):
            key1, key2 = jax.random.split(key)
            state = self._env.reset(key2)
            return state, state.obs, key1

        self._reset = jax.jit(reset, backend=self.backend)

        def step(state, action):
            state = self._env.step(state, action)
            info = {**state.metrics, **state.info}
            return state, state.obs, state.reward, state.done, info

        self._step = jax.jit(step, backend=self.backend)

    def seed(self, seed: int = 0):
        self._key = jax.random.PRNGKey(seed)

    def reset(self):
        self._state, obs, self._key = self._reset(self._key)
        # We return device arrays for pytorch users.
        return obs

    def step(self, action):
        self._state, obs, reward, done, info = self._step(self._state, action)
        # We return device arrays for pytorch users.
        return obs, reward, done, info

    @property
    def observation_size(self):
        return self._env.observation_size

    @property
    def action_size(self):
        return self._env.sys.nu

```
 and instantiated like so: 

`env = GymWrapper(env, seed=spec['seed'], backend='gpu')`

Appreciate the help!

btaba, 2023-10-26:
Hi @SumeetBatra thanks for the question, a few questions on our end:

1. Are you using the humanoid XML in the MJX repo, or is it different/modified?
2. Are you comparing to Brax `generalized` or another backend? 
3. Are you comparing Brax/MJX on the same mjcf? The XML in MJX is not the same as in Brax. Can you confirm that the number of constraints is the same for both simulators (joint/contact/limit/etc.)?
4. Do the Brax and MJX envs do the same number of physics steps per env step?

SumeetBatra, 2023-10-26:
Yes, it is the humanoid XML in the mjx repo unmodified. I should clarify -- the 200k fps result was collected on an older release of brax (v0.1.0), so I'm comparing mjx backend, which I'm assuming is `generalized` by default, to the `legacy_spring` version in the older release. 

You are right, the xml's are quite different. I see 17 bodies, 22 joints, 4 collision pairs in the mujoco xml. That's compared to 12 bodies, 10 joints, and 2 collision pairs for the brax xml. AFAICT the number of physics steps per control step is 8 in brax and 5 in mjx.

btaba, 2023-10-26:
Thanks, that's an important clarification. {`legacy_spring`} and {MJX, `generalized`} are different sets of physics backends. `legacy_spring` uses an iterative solver in maximal coordinates (violates joint constraints, needs spring stiffness/damping tuning, not robust to non-spherical non-identity inertias),  useful for RL jobs that don't require transfer of policies to the real world. v0.1.0 of Brax contained `generalized` in a `v2` folder, which was not the default physics backend.

 {MJX, `generalized`} physics backends uses Featherstone's, does not violate joint constraints, and is well tested for sim2real transfer, albeit more computationally expensive.  Your reported fps on the RTX 3090 is a 50% improvement over the number we report in colab on a V100 (assuming same physics steps per env steps, etc.).

My suggestion is to keep using brax spring/positional if you want to do pure RL and don't mind tuning the physics. Use `generalized` or MJX if you want a more robust simulator for robotics use-cases. Hope this helps for now

btaba, 2023-10-27:
Closing this, but feel free to re-open if there are more questions

closed 2023-10-27 by btaba

erikfrey, 2023-10-27:
Two other notes:

1) you can further tune MJX params if all you want is a stable simulation that trades physics fidelity for speed.  See these parameters here:

https://github.com/google-deepmind/mujoco/blob/main/mjx/mujoco/mjx/benchmark/model/humanoid/humanoid.xml#L17

Just now I tried lowering them to `iterations="2" ls_iterations="4"` -- the simulation was still stable, and still far better than the motion quality in Brax's `legacy_spring` pipeline, and improved steps/sec by 30-40% on my A100.

See this [performance tuning](https://mujoco.readthedocs.io/en/latest/mjx.html#performance-tuning) section of the MJX documentation for a few pointers.

2) Most of the time in an MJX physics step is spent in the constraint solver.  I'm currently tinkering with getting the [Newton solver](https://mujoco.readthedocs.io/en/latest/modeling.html#solver-settings) into MJX which may provide a healthy speedup.  And we have other optimizations slated after that, too.

SumeetBatra, 2023-10-27:
Thank you both for the really helpful comments, I'll spend some time tinkering around with these params. To provide some context, I am a PhD candidate doing robotics research, and while I do ultimately care about sim2real transfer, it is useful during the iteration phase to have maximum throughput, and so I think having configs on both ends of the spectrum would be really useful for researchy tasks. Glad to hear that the newton solver and other optimizations will make their way into the pipeline!