google-deepmind/mujoco #1136: direction of normal vector in mjContact

MiaoDragon, 2023-10-26:
Hi,

I'm a PhD student and I'm trying to use MuJoCo for a robotic project involving contacts. I'm looking for some help with the specifications of mjContact. I have read through the documentation, but haven't found any description of the direction of the contact normal vector. For instance, is the normal vector pointing from geom1 to geom2, or the other way? 

This is important for me to construct optimization problem involving the contacts. For example, if I want to construct a constraint to avoid penetration, I need to use the relative velocity of geom1 relative to geom2 in the direction of the normal vector. If the normal vector points from geom1 to geom2, then the relative velocity in the normal direction has to be negative to separate them. Otherwise, it should be positive. This is illustrated in the graph below:

[9ab5bcce-dc6d-40f4-a267-23531764f257.pdf](https://github.com/google-deepmind/mujoco/files/13180159/9ab5bcce-dc6d-40f4-a267-23531764f257.pdf)
In this graph, the relative velocity of geom1 w.r.t. geom2 needs to have negative projection on the normal vector to avoid penetration. Hence the direction of the normal vector matters.

v-r-a, 2023-10-26:
Hello, I am a MuJoCo user.

In the documentation, you may want to trace back from the last few lines of the [contact ](https://mujoco.readthedocs.io/en/latest/programming/simulation.html?highlight=MjContact#contacts)section. But the direction is not explicitly mentioned there. I think the discussion in issue #679 may be helpful to you.

I hope you also know the [mj_contactForce ](https://mujoco.readthedocs.io/en/latest/APIreference/APIfunctions.html#mj-contactforce) function.

Thank you.

MiaoDragon, 2023-10-26:
Thanks for the quick response! I'm aware of the contact section in the documentation. However, it didn't mention anything regarding the direction of the contact normal vector.
For the contactForce function, it seems that it didn't show how the force is associated with geoms. This doesn't seem to help if I need to explicitly construct a custom constraint on object body velocities.
Probably the documentation could be refined by adding descriptions of the direction of the normal vector w.r.t. geom1 or geom2 if there is such a restriction in the code. This may help clarify.

I have manually visualized the contact normals as arrows as shown in the following picture.
[image]
It seems that some contacts have body1 being workspace, while others have body2 as workspace.
This might be another issue, but the normal vectors don't seem to point perpendicular to the contact surface.
Any help would be appreciated!

v-r-a, 2023-10-26:
> Probably the documentation could be refined by adding descriptions of the direction of the normal vector w.r.t. geom1 or geom2 if there is such a restriction in the code. This may help clarify.

I had answered the same question for myself earlier in issue #679 as follows:

> For question 1, I guess the normal points away from con.geom1.
> 
> For question 2, I guess the frame belongs to the body of con.geom1 except when it is the world body. (I am pursuing this to retrieve the direction of friction forces).

I will let the developers/ other experts answer it definitively.

MiaoDragon, 2023-10-26:
Thanks for the reply.
I have plotted something to visualize the contact normals and positions. I assume the geom::mat field stores in row-major order, and the z axis specifies the pointing direction of arrow geometry object. Below is the plot.
[image]
The workspace refers to the table. So from this I think the mujoco behavior is pointing from body1 to body2.

Nevertheless, I think the documentation should include this and enforce this order for related functions. For example, the function [mj_addContact](https://mujoco.readthedocs.io/en/latest/APIreference/APIfunctions.html#mj-addcontact) doesn't seem to care about the normal vector direction that user provides. This could lead to issues of inconsistency. Still hope that the developers could response to this.

Thanks!

yuvaltassa, 2023-11-04:
Ya this is not well documented. If you go [here](https://mujoco.readthedocs.io/en/latest/XMLreference.html#contact-pair) you can see the sentence

> he contact force vector computed by the solver and stored in mjData.efc_force points from the first towards the second geom by convention.

But that is hardly discoverable...

@MiaoDragon you know that MuJoco has built-in contact force visualisation, right? This is covered in the [tutorial](https://github.com/google-deepmind/mujoco#getting-started)

commit 8a5e2b19 (2023-11-06): Improve `mjContact` documentation. Fixes #1136

PiperOrigin-RevId: 579836970
Change-Id: Ife8dd73eecdbdaacecc1e48460cf6efcf4f7b487
closed 2023-11-06 by copybara-service[bot]