google-deepmind/mujoco #1137: CMake: why are system packages ignored?

tmplt, 2023-10-26:
I am packaging v3.0.0, including Python bindings, for NixOS. During build of MuJoCo dependencies are fetched by commit revision and system packages are ignored. I find this unconventional. Is this a combination of end-user convenience and fetching API limitation, or are there dependencies which are found in off-shoot branches? That is, can I assume that each fetch is for a released dependency version, and thus set `USE_SYSTEM_PACKAGE ON`?

traversaro, 2023-10-26:
You may be interested in https://github.com/google-deepmind/mujoco/pull/937 . Also some other patches we have on conda-forge can be interesting for you: https://github.com/conda-forge/mujoco-feedstock/tree/main/recipe .

saran-t, 2023-10-26:
@tmplt physics engines can be really sensitive to small changes, so to keep some control over the behaviour for each released version we strongly prefer to pin exact versions of all dependencies and statically link. 

I realise that it's unconventional, but in the ideal world I'd prefer it if distros just package our prebuilt binaries as distributed over GitHub. Dependency symbols are all appropriately hidden so there shouldn't be much concern over conflicts with system packages. (It seems like this will be too much to ask for most distros though...)

tmplt, 2023-10-26:
@saran-t thanks for the quick response. That makes sense. A lot of the dependencies seem numerical after all. 

> we had to modify dependency code

Thanks for the heads-up. I'll wrangle with the pins, then.

closed 2023-10-26 by tmplt

fferri, 2025-07-01:
> Dependency symbols are all appropriately hidden so there shouldn't be much concern over conflicts with system packages. (It seems like this will be too much to ask for most distros though...)

not the case on macOS. libmujoco.dylib exports tinyxml2 symbols, which leads to a crash when also the client program uses system's tinyxml2.